react-native-camera/react-native-camera #1432: Multiple Camera Preview not working

kosintop, 2018-04-03:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. Minimum code
[image]

### Expected behaviour
Both camera preview should working normally

### Actual behaviour
Only the second one is working normally, the first one will show a freezed frame 

### Environment
- **Node.js version**: 9.0.0
- **React Native version**: 0.50.0
- **React Native platform + platform version**:  Android

### react-native-camera
**Version**:  "git+https://git@github.com/react-native-community/react-native-camera"

sibelius, 2018-04-03:
could u check if this works with expo camera https://github.com/expo/expo?

kosintop, 2018-04-04:
@sibelius I tried to render two camera with expo camera and it crash with java exception

`E/AndroidRuntime(10692): java.lang.RuntimeException: java.io.IOException: setPreviewTexture failed
E/AndroidRuntime(10692):        at com.google.android.cameraview.a.c(Camera1.java:176)`

Does this mean react-native-camera will not be able to render two camera too?

sibelius, 2018-04-04:
This means that we need a pull request to fix this

jgfidelis, 2018-04-04:
In Android it is possible to have only one CameraView instantiated. I don't know why specifically, but that is how it is, the second one does not mount. Maybe a limitation in our camera view class: https://github.com/react-native-community/cameraview

nhayfield, 2018-07-20:
Is there a solution for this?

sibelius, 2018-07-20:
maybe the problem is here https://github.com/react-native-community/react-native-camera/blob/master/src/RNCamera.js#L447

try to do something like this PR

https://github.com/react-native-community/react-native-linear-gradient/pull/269

LautaroMir93, 2019-04-04:
Any updates on this?

prodoxx, 2019-05-19:
I'm still having this issue. I'm getting a ```setPreviewTexture failed``` exception error when I try to remove the view with the camera and change to another view. It turns out the problem is because I'm using two cameras instances in different areas of my app. I'm using it in two different screens and not on the same screen. If I disable the camera from the other screen and try again, it works fine. 

I don't know how to turn off the camera from the screen the user is not viewing. I have the latest version of RNCamera installed (2.9.0). 

Any idea on how to solve this?

closed 2019-07-25 by stale[bot]