react-native-camera/react-native-camera #1434: RNCamera seems to disable Vibration.vibrate()

peterjuras, 2018-04-04:
Reopened from #397 
 
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. Mount a RNCamera component
2. Call `require('react-native').Vibration.vibrate()` through a button click or similar

### Does it work with Expo Camera?
I did not test with Expo Camera

### Expected behaviour
The device should vibrate

### Actual behaviour
The device does not vibrate

### Environment
- **Node.js version**: 9.8.0
- **React Native version**: 0.49.3
- **React Native platform + platform version**: iOS 11.1 & Android 8.0 (probably others as well)

### react-native-camera
**Version**: 1.0.3

ma-pe, 2018-04-05:
In my experience it works on both platforms as long as you don't have an active recording.

iOS doesn't allow vibration while recording video with audio on. Maybe there is a possibilty when it comes to audio channel mixing since the vibration on iOS is handled as a system sound.

Though it is possible to vibrate while the video recording is muted.

peterjuras, 2018-04-06:
Thanks for your quick reply.

Sadly for me it works on neither platform. I'm not recording video and captureAudio ist left to the defaultValue of false, so it should be muted. This is the code I'm using:

```js
// ...
onQRRead = ({ type, data } = {}) => {
  // Not working
  Vibration.vibrate();

  // Working
  this.props.navigateAwayAndUnmount();
  setTimeout(() => Vibration.vibrate(), 250);
};
// ...
<RNCamera
  autoFocus
  style={styles.cameraView}
  type={RNCamera.Constants.Type.back}
  barCodeTypes={[RNCamera.Constants.BarCodeType.qr]}
  flashMode={
    flash
    ? RNCamera.Constants.FlashMode.torch
    : RNCamera.Constants.FlashMode.off
  }
  onBarCodeRead={this.onQRRead}
  permissionDialogTitle="Permission to use camera"
  permissionDialogMessage="ORDA Setup needs access to camera to read QR codes"
/>
// ...
```

jgfidelis, 2018-04-06:
https://facebook.github.io/react-native/docs/vibration.html

Docs says you need to pass a number to `.vibrate()`. Can you try `.vibrate(1000)`?

jgfidelis, 2018-04-06:
maybe related to: https://github.com/facebook/react-native/issues/8955 ?

I just tried on the example app to call Vibration.vibrate() after pressing takePicture button and it vibrated.

So having the camera open is not the main problem. I do not think this is exactly related to the camera.

peterjuras, 2018-04-12:
You're right, this might be the culprit. I'll be monitoring the linked issue then, thank you!

dFelinger, 2020-06-17:
Use captureAudio={false} on RNCamera if you still have this issue.

frangeris, 2020-08-31:
@dFelinger that way works, do you know the explanation why its failing?

frangeris, 2020-09-02:
@jgfidelis This is still happening

The only way the get vibrate api working is setting `captureAudio={false}`. as @dFelinger said, so if you need videos with audio and support for vibrate, things gets difficult

So I tried to disable to get video with audio but changing `captureAudio` based on state is a buggy, I'm getting a lot of exceptions when using state to enable or disable audio:

```
Unhandled promise rejection [Error: You attempted to set the key `mute` with the value `true` on an object that is meant to be immutable and has been frozen.]
```

```js
<RNCamera
    {...!isRecording && { captureAudio: false }}
    ref={camera}
    // ....
/>
```

What other approach could we have?

zoolyka, 2020-10-12:
Can confirm that this issue also present with latest Expo (SDK39) too.

closed 2021-01-13 by stale[bot]