react-native-camera/react-native-camera #1435: Request for Barcode scan area support

kosintop, 2018-04-04:
I have been looking for ways to limit barcode scan area and have notice that RNCamera already have the  needed structure to support this feature on android (not sure about iOS)

In the [BarCodeScannerAsyncTask.java line#62](https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/tasks/BarCodeScannerAsyncTask.java#L62), the PlanarYUVLuminanceSource contructor which take in the previewed frame can take parameter top,left,width,and height which in effect will cropped the frame and will only scan for barcode in that specified area

I have tried roughly edit the code, set the mentioned params, and create a rectangle scan area in react side to see the result. It seems to be working but the area seems to be a little bit off which might be because the frame can get skewed depends on aspect ratio

this features has been request a few time #962, #815, and there is also a pull request #874 (but for RCT camera and does not get merged)

I'm not sure how the whole thing work so I would like to request this feature to be implemented

tidydee, 2018-04-27:
Been looking at this myself. Would love to see this feature come to light.

winadiw, 2018-09-24:
Any follow up on this? Really need this feature..

n1ru4l, 2018-09-24:
@winadiw Feel free to send a pull request that addresses this issue.

sibelius, 2019-03-25:
check this https://github.com/react-native-community/react-native-camera/pull/1852

closed 2019-06-07 by stale[bot]