react-native-camera/react-native-camera #1436: RNCamera view on android in portrait mode is squished.

mjaffa-figure, 2018-04-04:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using

*RNCamera* Mater branch  

### Steps to reproduce
1. Put an RNCamera view in a portrait mode only app. See the preview is squished. It assumes that you want to take the photo in landscape. Cause if I turn my device with portrait mode locked, the picture doesn't look squished.
2.
3.

### Does it work with Expo Camera?
RNCamera is based on Expo Camera, if this does not work with Expo Camera it won't work with RNCamera as well.
You should open an issue there as well, so we can cooperate in a solution.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**:
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

SimonErm, 2018-04-04:
Can you post a Screenshot?

jgfidelis, 2018-04-05:
Does this happen on version 1.0.3?

mjaffa-figure, 2018-04-05:
Yes, I was on 1.0.1. Did an update with npm update react-native-camera. It updated it to 1.0.3. Noticed in Portrait mode of phone it was squished. Took a picture. Then on next screen displayed that picture which it was rotated 90 degrees and photo did not look squished. 

When I set this option in the takePictureAsync I notice the image is not rotated 90 degrees: fixOrientation: true

And I have tried this on master as well. The camera preview screen always thinks is in Landscape mode.

mjaffa-figure, 2018-04-05:
Working on getting screenshots, just got to get approval to screenshot where we are using it in our unreleased app. Or if I have time to get a sample project made.

jgfidelis, 2018-04-05:
Are you on Android or iOS? fixOrientation is Android only. In iOS forceUpOrientation: true, might help.

mjaffa-figure, 2018-04-05:
This is happening on Android. iOS is working perfectly.

mjaffa-figure, 2018-04-05:
Here is a sample project where I am seeing this in portrait mode. 
[Squished.zip](https://github.com/react-native-community/react-native-camera/files/1880920/Squished.zip)

jqn, 2018-04-05:
@mjaffa-figure checkout this issue https://github.com/react-native-community/react-native-camera/issues/1369

mjaffa-figure, 2018-04-05:
I tried setting the aspect ratios, I am a Note 8. Sample project I attached above will show the squished face.

mjaffa-figure, 2018-04-05:
Yup, I modified the sample code, it does nothing and did log statements to make sure it was getting to the code.

mjaffa-figure, 2018-04-05:
Here is the sample project with those changes you suggested I make. 
[Squished 2.zip](https://github.com/react-native-community/react-native-camera/files/1881116/Squished.2.zip)

sibelius, 2018-04-05:
it is better to upload on github, instead of zip

jqn, 2018-04-05:
Github repo is definitely better. I can't really run your example @mjaffa-figure. In any case is usually better to simplify what you are trying to debug, with that said I would remove all the extra stuff and just render a screen with just the camera.

SimonErm, 2018-04-05:
Got it runnig with master-branch and ratio=1:1 on OnePlus3T, i am not sure if that work on devices with smaller screens, because there is no supported previewsize maybe, if you use the preview just on the half screen and that's the reason why it squishes with bigger ratios. If you use master and remove your example text view, it will be cropped to a working ratio. If you want to use the highest ratio, you should add the Button as a child of RNCamera.

mjaffa-figure, 2018-04-05:
1:1 works for me.

closed 2018-04-05 by mjaffa-figure

ma-pe, 2018-04-17:
@mjaffa-figure I have the same issue. The _preview_ is squished on android.
1:1 doesn't fix the issue for me. 

There was a property for this in RCTCamera: https://github.com/react-native-community/react-native-camera/blob/master/docs/RCTCamera.md#aspect