react-native-camera/react-native-camera #1443: error: cannot find symbol new RNCameraPackage()

Aashir1, 2018-04-07:
I simply followed the steps that mentioned in react-native-camera documentation but when run command react-native run android it gives error that.

D:\Aashir\Course\React Native\Cemra Access Task\react-native-boilerplate\android\app\src\main\java\com\boilerplate\MainApplication.java:6: error: package org.reactnative.camera does not exist
import org.reactnative.camera.RNCameraPackage;
                             ^
D:\Aashir\Course\React Native\Cemra Access Task\react-native-boilerplate\android\app\src\main\java\com\boilerplate\MainApplication.java:30: error: cannot find symbol
          new RNCameraPackage(),
              ^
  symbol: class RNCameraPackage
2 errors
:app:compileDebugJavaWithJavac FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 28.577 secs
Could not install the app on the device, read the error above for details.
Make sure you have an Android emulator running or a device connected and have
set up your Android development environment:
https://facebook.github.io/react-native/docs/getting-started.html
PS D:\Aashir\Course\React Native\Cemra Access Task\react-native-boilerplate>

Aashir1, 2018-04-07:
I am not only facing this issue also see it [#2](https://github.com/react-native-community/rncamera-example/issues/2)

jgfidelis, 2018-04-07:
Your linking is wrong... Hard to know in what step you made a mistake. Describe briefly what you did.

closed 2018-04-07 by sibelius

ChandniSharma, 2018-09-24:
I am also facing this issue with SoundModulePackage. 
Can any one has a solution?

arasemami, 2018-09-25:
in the android/app/build.gradle at dependencies you must add this line to read package. 

`dependencies { 
    ...............
    compile project(':react-native-sound-player')
}`

it will be working ;)

wizzard94, 2019-02-02:
Take a look at step 4 in the documentation
https://github.com/react-native-community/react-native-camera#android