react-native-camera/react-native-camera #1447: crashed when permission is off

ZiXYu, 2018-04-08:
### Which implementation are you using

RNCamera

### Steps to reproduce
1. import react-native-camera like
```
 <Camera
          style={styles.container}
          captureAudio={false}
          notAuthorizedView={<Text>
            no authorized! Please try again!
          </Text>}
          onMountError={() => {
            console.log('on mount error!')
          }}
          onBarCodeRead={this.barCodeRead}
          aspect={Camera.constants.Aspect.fill} />
```
2. on iOS, go to the setting page and turn off the permission
3. go back to camera, it will crash instead of show the notAuthorizedView
4. and found the crash log on xCode
```
Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[AVCaptureMetadataOutput setMetadataObjectTypes:] Unsupported type found - use -availableMetadataObjectTypes'
*** First throw call stack:
(0x182d5dd04 0x181fac528 0x1886b0828 0x102ce8ddc 0x10331149c 0x10331145c 0x103320110 0x1033149a4 0x103321104 0x103328100 0x182986fd0 0x182986c20)
libc++abi.dylib: terminating with uncaught exception of type NSException
(lldb) 
```

### Expected behaviour
Camera should show the notAuthorizedView, should not crash immediately.

### Actual behaviour
It crashed.

jgfidelis, 2018-04-09:
This is RCTCamera.

n1ru4l, 2018-09-24:
RCTCamera is deprecated, please try RNCamera and open a new issue if you still experience this issue.

closed 2018-09-24 by n1ru4l