react-native-camera/react-native-camera #1448: I get a picture?

yz-xlame, 2018-04-09:
export default class App extends Component {
  render() {
    return (
      <View style={styles.container}>
        <RNCamera
            ref={ref => {
              this.camera = ref;
            }}
            style = {styles.preview}
            flashMode={RNCamera.Constants.FlashMode.auto}
            permissionDialogTitle= {'Permission to use camera'}
            permissionDialogMessage= {'We need your permission to use your camera phone}
        />
        <View style={{flex: 0, flexDirection: 'row', justifyContent: 'center',}}>
          <TouchableOpacity
              onPress={this.takeRecoding.bind(this)}
              style = {styles.capture}
          >
              <Text style={{fontSize: 14}}> SNAP </Text>
          </TouchableOpacity>
        </View>
      </View>
    );
  }

  takeRecoding = async function() {
    if(this.camera){
      const videoOptions = {
        quality: RNCamera.Constants.VideoQuality.480p,
        maxDuration: 3,
      }
      const data = await this.camera.recordAsync(videoOptions)
      console.log(data)
    }
  };

}

yz-xlame, 2018-04-09:
I'm not sure if it is used like this, return a picture

yz-xlame, 2018-04-09:
recordAsync and takePictureAsync Promise does not return,I don't know if I got a return from there

SimonErm, 2018-04-09:
use 
try{
  const data=await...
}catch(error){
  console.log(error);
}
and try if it throws an error.

yz-xlame, 2018-04-10:
Nothing throws an error and always doesn't know where to throw the picture path from