react-native-camera/react-native-camera #1449: react native camera captured picture not save in app cache

Aashir1, 2018-04-09:
I am using react-native-camera in my react-native app and everything works well, however when I take a photo the picture should be saved on my device, I addressed the mention location where my picture saves but there is no any image. How can I access capture image? I m using takePictureAsync.

yz-xlame, 2018-04-09:
recordAsync([options]): Promise
The promise will be fulfilled with an object with some of the following properties:

uri: returns the path to the video saved on your app's cache directory.

iOS codec: the codec of the recorded video. One of RNCamera.Constants.VideoCodec

They say so

jgfidelis, 2018-04-09:
It is saved on your cache directory. Do you have all correct permissions? If you want to save it somewhere else, like the Camera roll, you should use other libraries.

jgfidelis, 2018-04-09:
Are you using RNCamera, RCTCamera...? Which version?

Please explain your problem better with a code snippet if possible.

Aashir1, 2018-04-10:
"react": "16.3.0-alpha.1",
"react-native": "0.54.2",
"react-native-camera": "^1.0.3".

I am using RNCamera and with all permissions that mentioned in the react-native-camera documentation.

Thank you, @jgfidelis Camera roll works fine for my use case.

closed 2018-04-10 by jgfidelis

josuesossou, 2018-06-21:
@AndrewHenderson  I want to do the same and even if it's possible not to save it in chache

AndrewHenderson, 2018-06-21:
@josuesossou I'm ended up using [React Native FS](https://github.com/itinance/react-native-fs) to remove the file once I was done with it. 
```jsx
import { RNCamera } from 'react-native-camera';
import RNFS from 'react-native-fs';

takePhoto = async () => {
  const data = await this.camera.takePictureAsync();
  // Do what you need with the image and then…
  RNFS.unlink(data.uri); // Remove image from cache
}
```

josuesossou, 2018-06-21:
@AndrewHenderson thank you so much, that was a great help!

ramipaleo, 2018-09-28:
Hi @Aashir1 , How did you use Camera roll? I have all permissions that mentioned in the react-native-camera, but It show me error of access denied

akbarmaulan4, 2018-11-27:
same issue, I'm using RNCamera, when get data uri I try to save to my local storage with react-native-fs,