react-native-camera/react-native-camera #1450: Don't want to save image on device

remisaado, 2018-04-09:
I am using react-native-camera in my react-native app and everything works well, however when I take a photo the picture saves on my device which I do not want, the image is supposed to be sent to an api but NOT saved on my device, this is a suggestion for something you could implement to the library, unless it already exists?

Also if you know of any way to not have the photo saved on device please check out my question on [Stackoverflow](https://stackoverflow.com/questions/49694246/react-native-camera-dont-want-to-save-images).

```
render() {
return(
  <View style={styles.container}>
    <Camera
      style={styles.camera}
      ref={ref => {this.camera = ref;}}
      type={Camera.constants.Type.back}
      flashMode={Camera.constants.FlashMode.on}
      orientation={Camera.constants.Orientation.landscapeLeft}>
      <TouchableOpacity onPress={this.takePicture.bind(this)} style={{zIndex: 100}}>
        <View style={styles.cameraButton}></View>
      </TouchableOpacity>
    </Camera>
  </View>
  );
}

takePicture()
{
  this.camera.capture()
  .then((data) => this.sendData(data))
  .catch(err => console.error(err));
}
```

jgfidelis, 2018-04-09:
Why can't you save it on your cache directory, what is the problem with that? The picture is saved there and will be erased when the cache is cleaned.

And like I answered in SO, to send it to an api, the best way is getting the base64 string and sending in your request.

Not saving the image today is not possible, it would be a feature request.

remisaado, 2018-04-09:
But it's not being deleted currently, it's being saved in my image gallery on my device, how can I get it to save on cache only?

I'm already sending it to an api, all of that is working, I just don't want the image to be saved in the gallery on the device.

jgfidelis, 2018-04-09:
Oh, I see, you are using RCTCamera. Then I believe it is saved to the device by default.

Read the docs: https://github.com/react-native-community/react-native-camera/blob/master/docs/RCTCamera.md

Use the prop `captureTarget`, passing the value `Camera.constants.CaptureTarget.temp`.
So:
```
<Camera
...
captureTarget={Camera.constants.CaptureTarget.temp}
...
</Camera>
```

And see if it works.

Or, consider updating your code to use RNCamera, which is now the officially supported implementation on this package.

remisaado, 2018-04-09:
@jgfidelis Awesome it worked! Thanks.

whwz163, 2018-10-31:
> Oh, I see, you are using RCTCamera. Then I believe it is saved to the device by default.
>
> Read the docs: https://github.com/react-native-community/react-native-camera/blob/master/docs/RCTCamera.md [...]

which props in RNCamera same as captureTarget