react-native-camera/react-native-camera #1451: App alway crash when use RNCamera

vanhungoz, 2018-04-09:
i using RNCamera in my app with some information 
react-native :0.52.0
"react-native-camera": "^1.0.3",
when i use " import Camera from 'react-native-camera';". it take a error "Permission denied".
[image]
-when i use "import { RNCamera } from 'react-native-camera';" . it make my app crash whenever i navigate to this screen.
anyone saw this error before.

sibelius, 2018-04-09:
You need a permission to storage the image when taking photo

only after getting this permission that you can show the camera

closed 2018-04-09 by sibelius

vanhungoz, 2018-04-09:
what information do you want? maybe i can show it for you ?

vanhungoz, 2018-04-09:
"react-native-camera": "^1.0.3",
    "react-native": "0.52.0",
import {RNCamera} from 'react-native-camera';

vanhungoz, 2018-04-09:
it make crash when i use import {RNCamera} from 'react-native-camera';
i can't log anything before it crash

sibelius, 2018-04-09:
Try to use adb logcat

vanhungoz, 2018-04-09:
adb logcat show me error is :"ErrorProcessor: Caused by: com.google.android.apps.gsa.shared.exception"

sibelius, 2018-04-09:
we need the full stacktrace

vanhungoz, 2018-04-09:
yes. i'll send to you soon

vanhungoz, 2018-04-09:
can i set `compile (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
        compile 'com.android.support:exifinterface:27.+'
//        compile ("com.google.android.gms:play-services-vision:12.0.1")
    }`

vanhungoz, 2018-04-09:
without 'com.google.android.gms:play-services-vision:12.0.1'

vanhungoz, 2018-04-09:
in android . error is "Didn't find class "com.google.android.gms.vision.face.FaceDetector"

vanhungoz, 2018-04-09:
may i use this lib without FaceDetector in android ?

jgcmarins, 2018-04-09:
yes, @vanhungoz 
try to follow docs, please

vanhungoz, 2018-04-10:
hi @jgcmarins  ,in android it work perfectly when i install lib with GVM. but whenever i install with  'com.google.android.gms:play-services-vision:12.0.1'.  it make app crash with logcat "Didn't find class "com.google.android.gms.vision.face.FaceDetector". 
i try to read README but according to README i just saw how to remove GVM in IOS, i can't see anything about ANDROID .

jgcmarins, 2018-04-11:
I guess all `play-services` must be on same version

vanhungoz, 2018-04-11:
now, i am installing GVM with no ideal to use it in my app.lol

sibelius, 2018-04-11:
gvm is not required on ios

jgfidelis, 2018-04-12:
1 - GVM is required on Android at all times.
2 - It is optional on iOS.

Are you still having any problem?

vanhungoz, 2018-05-30:
thanks @jgfidelis . now it work for me.