react-native-camera/react-native-camera #1453: Flash is not working in case of VideoRecording in both IOS and android

rishaChoubey, 2018-04-09:
### Steps to reproduce
1. Tap on add video.
2. Tap on start video.
3. Tap on flash button.

### Expected behaviour
Flash should be enabled while recording the video

### Actual behaviour
Flash is not working while recording the video

### Environment
- **Node.js version**:5.6.0
- **React Native version**:0.50.3
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: ^1.0.2

jgfidelis, 2018-04-09:
What type of flash are you using?

rishaChoubey, 2018-04-09:
@jgfidelis : 

I am using react-native-camera flash

jgfidelis, 2018-04-09:
no, what is the value of the prop `flashMode` you are using?

Do you have the code snippet of your camera in your `render()` function?

devapro, 2018-08-07:
for android added <uses-permission android:name="android.permission.FLASHLIGHT"/> to manifest.
now it work well on master branch

closed 2018-08-07 by devapro

SiyabongaNzulwana, 2019-12-06:
Hi guys, I need help with as well. I am having a that is responsible for recording a video, I want to be able to toggle the flashMode. please help.

```
 <RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          style={styles.preview}
          type={RNCamera.Constants.Type.back}
          flashMode={flashMode}
          androidCameraPermissionOptions={{
            title: 'Permission to use camera',
            message: 'We need your permission to use your camera',
            buttonPositive: 'Ok',
            buttonNegative: 'Cancel',
          }}
        />

<TouchableOpacity
                    style={styles.flashIcon}
                    onPress={() =>
                      this.setState({
                        flashMode: flashMode === 'off' ? 'auto' : flashMode === 'auto' ? 'on' : 'off',
                      })
                    }>
                    <Icon
                      name={flashMode === 'off' ? 'flash-off' : flashMode === 'on' ? 'flash-on' : 'flash-auto'}
                      size={18}
                      color="#fff"
                      style={{position: 'absolute', marginTop: 1,  color: '#FFF',}}
                      type="material"
                    />
                  </TouchableOpacity>
```

SiyabongaNzulwana, 2019-12-06:
> no, what is the value of the prop `flashMode` you are using?
> 
> Do you have the code snippet of your camera in your `render()` function?

```
<RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          style={styles.preview}
          type={RNCamera.Constants.Type.back}
          flashMode={flashMode}
          androidCameraPermissionOptions={{
            title: 'Permission to use camera',
            message: 'We need your permission to use your camera',
            buttonPositive: 'Ok',
            buttonNegative: 'Cancel',
          }}
        />

<TouchableOpacity
     style={styles.flashIcon}
      onPress={() =>
        this.setState({
        flashMode: flashMode === 'off' ? 'auto' : flashMode === 'auto' ? 'on' : 'off',
          })
        }>
         <Icon
           name={flashMode === 'off' ? 'flash-off' : flashMode === 'on' ? 'flash-on' : 'flash-auto'}
           size={18}
           color="#fff"
            style={{position: 'absolute', marginTop: 1,  color: '#FFF',}}
            type="material"
            />
   </TouchableOpacity>
```
Please assist, thats my code

PR #1731 (2018-08-08): Fix ios flash
fix #1453 , #1634 
fix "on" and "auto" flash mode for ios