react-native-camera/react-native-camera #1454: Broken Framework Search Paths, need to manually edit them every time we run yarn install

VicFrolov, 2018-04-09:
### Which implementation are you using
RNCamera

### Steps to reproduce
1. run yarn install
2. try to build react app
3. get red warning error

### Expected behaviour
Proper compilation and paths

### Actual behaviour
I need to go into Xcode, go to libraries, RNCamera, build settings, and go through the search paths which end in a */ios and delete one of the `../`

For example, Framework Search Path is `$(SRCROOT)/../../../ios` when it should be `$(SRCROOT)/../../ios`

### Environment
- **Node.js version**: 9.5.0
- **React Native version**: "^0.52.0"
- **React Native platform + platform version**:  iOS 11.0

### react-native-camera
**Version**: using "master"

"react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera.git"

sibelius, 2018-04-09:
Send a pr please

nhu, 2018-04-10:
Hi,
I stumbled across a similar issue related to the search paths. For me it was an issue with expanding the search paths because they are too long.

I guess these issues are both related to one of the changes listed below: 

- https://github.com/react-native-community/react-native-camera/commit/31143ca29557e4077803d98524d43a1095fecd7c#diff-ad844da7be9b2c317fe08e932d2e0720
- https://github.com/react-native-community/react-native-camera/commit/4496c3217195853a36c261415f126140ddebbcc4#diff-ad844da7be9b2c317fe08e932d2e0720

Maybe it should be considered to replace the search path which is currently formulated as 
`$(SRCROOT)/../../../ios/**`
to `${BUILT_PRODUCTS_DIR}` if it really points to the same directory.

For me it resolved the build issues.

My iOS/xCode skills are very limited. For this reason I will only make a PR if you really want it and if you have double checked that there are no side effects given by replacing this paths. I currently can't really tell if this only works on my machine ;-)

Cheers

jgfidelis, 2018-04-10:
@nhu, thanks for the tip, I will investigate during the weekend!

@VicFrolov How is your project folder structure?

Because I was the one who changed from `$(SRCROOT)/../../ios` to `$(SRCROOT)/../../../ios`.

I changed because there was an error for our project.

Our structure is, for an app called AwesomeApp:

AwesomeApp/native_modules/react-native-camera/ios. This should be your $(SRC_ROOT). This is where the RNCamera xcode project is. So the path ../../../ points to AwesomeApp, and ../../../ios points to AwesomeApp's xcode project, where our frameworks are located.

Why does ../../ work for you? Shouldn't it point to native_modules?

I am just trying to understand why our structures are different. Thanks!

jgfidelis, 2018-04-10:
And you can always set an post-install script so you do not need to manually do it after every yarn install.

jgfidelis, 2018-04-10:
Hm... weird. Are you using Face Detection?

Looking at your path makes me believe that `../../` points to `node_modules`. Or am I missing something? There shouldn't be a `ios` folder at `node_modules`, but it should be in its parent folder, that is why `../../../ios` should work.

Can you go to your App's xcode project directory and type `pwd`, do the same at the RNCamera's xcode project directory and paste both results here?

jgfidelis, 2018-04-10:
@nhu, sorry, but your tip did not work

VicFrolov, 2018-04-10:
I am not using Face Detection.

MyApp.xcworkspace pwd: `/Users/MyUser/git/MyApp/ios/MyApp.xcworkspace`
RNC pwd: `/Users/MyUser/git/MyApp/node_modules/react-native-camera/ios/RNCamera.xcodeproj`

I see what you're saying now, this is strange, but the paths seem correct.

jgfidelis, 2018-04-11:
I see that you are using xcworkspace, so you installed via Cocoapods, am I correct? I am sure this is related with this, but I am not familiar enough with RN + Cocoapods to help you correctly. We avoid cocoapods entirely at our projects.

jgfidelis, 2018-04-11:
A solution would be to have both: `$(SRCROOT)/../../ios` and `$(SRCROOT)/../../../ios`, I don't know if its the best one though.

VicFrolov, 2018-04-12:
We have both xcworkspace and xcodeproj, but we only use react-native link, and yarn install. No cocoapods as they were causing issues. the paths are the same for both xcodeproj and xcworkspace

jgfidelis, 2018-04-12:
I see that most rn bridges use `${SRCROOT}/../../../ios`. Don't know why the {} though, wonder if there is any difference.

I think the solution is have both ../../../ios and ../../ios then, don't know why some people have this problem.

closed 2018-04-15 by jgfidelis

VicFrolov, 2018-05-03:
Hey just wanted to follow up and say this doesn't fix the problem. For some reason, it tries the ../../ first, and fails, but if I delete that, or place the ../../ first, it works.

dave-irvine, 2018-07-19:
I'm also facing the

> Argument list too long: recursive header expansion failed at `<snip>/node_modules/react-native-camera/ios/../../../ios/Pods/boost-for-react-native/boost/sort/spreadsort`.

problem when using CocoaPods. I've tried both installing `react-native-camera` via CocoaPods, and without (other dependencies still in CocoaPods) and still face the issue.

Interestingly, if I open the RNCamera xcode project, and completely delete the `FRAMEWORK_SEARCH_PATHS` and `HEADER_SEARCH_PATHS` values, everything builds happily and all works.

ducpt2, 2018-07-23:
@dave-irvine follow you and still can't work. Does anyone solve this problem, Pls?

VicFrolov, 2018-07-24:
Unfortunately, still waiting for a fix, and anytime I build on Xcode, I am required to update the search paths :/ 

I wonder if it matters how you start a react native project, as different paths work for different people, with identical file structures.

dave-irvine, 2018-07-24:
@VicFrolov I think it just matters how much "stuff" i.e other cocoapod dependencies can be found in the ios/* directory.

When you update the search paths, have you tried deleting all entries, not just the ones mentioning the ios/ directory? If so, does it still build? It does for me, so I'm wondering what the point of the entries is at all.

PR #1481 (2018-04-15): fix(search-paths): remove unnecessary search paths and add missing one
fix #1454