react-native-camera/react-native-camera #1458: Out of memory/crash when not in debugger

DevAlien, 2018-04-11:
### Which implementation are you using

*RNCamera

### Steps to reproduce
1. Created a new screen with the camera, using the remote debugger I have no issues and works perfectly. When I disable the debugger it takes forever to load the screen and it goes in out of memory.

### Does it work with Expo Camera?
Don't know

### Expected behaviour
It should not crash when the debugger is off

### Actual behaviour
The app should NOT crash/out of memory when the debugger is off

### Environment
- **Node.js version**: 9.3.0
- **React Native version**: 0.54.2
- **React Native platform + platform version**: iOS11.0, Android 8.0

### react-native-camera
**Version**: npm version

sibelius, 2018-04-11:
could u post the stacktrace?

DevAlien, 2018-04-11:
@sibelius the stack trace of the out of memory?

```
04-11 16:51:04.701 11922 12001 E ReactNativeJS: Error: Out of memory
04-11 16:51:04.701 11922 12001 E ReactNativeJS: 
04-11 16:51:04.701 11922 12001 E ReactNativeJS: This error is located at:
04-11 16:51:04.701 11922 12001 E ReactNativeJS:     in Camera
04-11 16:51:04.701 11922 12001 E ReactNativeJS:     in RCTView
04-11 16:51:04.701 11922 12001 E ReactNativeJS:     in RCTView
04-11 16:51:04.701 11922 12001 E ReactNativeJS:     in Record
04-11 16:51:04.701 11922 12001 E ReactNativeJS:     in ThemeProviderHOC
04-11 16:51:04.701 11922 12001 E ReactNativeJS:     in Provider
04-11 16:51:04.701 11922 12001 E ReactNativeJS:     in MyProvider
04-11 16:51:04.701 11922 12001 E ReactNativeJS:     in _class2
04-11 16:51:04.701 11922 12001 E ReactNativeJS:     in RCTView
04-11 16:51:04.701 11922 12001 E ReactNativeJS:     in _class
04-11 16:51:04.701 11922 12001 E ReactNativeJS:     in RCTView
04-11 16:51:04.701 11922 12001 E ReactNativeJS:     in RootSiblingsWrapper
04-11 16:51:04.701 11922 12001 E ReactNativeJS:     in RCTView
04-11 16:51:04.701 11922 12001 E ReactNativeJS:     in AppContainer
04-11 16:51:04.705 11922 12001 I ReactNativeJS: 'ref', null
04-11 16:51:04.706 11922 12001 E ReactNativeJS: Error: Out of memory
04-11 16:51:04.706 11922 12001 E ReactNativeJS: 
04-11 16:51:04.706 11922 12001 E ReactNativeJS: This error is located at:
04-11 16:51:04.706 11922 12001 E ReactNativeJS:     in Camera
04-11 16:51:04.706 11922 12001 E ReactNativeJS:     in RCTView
04-11 16:51:04.706 11922 12001 E ReactNativeJS:     in RCTView
04-11 16:51:04.706 11922 12001 E ReactNativeJS:     in Record
04-11 16:51:04.706 11922 12001 E ReactNativeJS:     in ThemeProviderHOC
04-11 16:51:04.706 11922 12001 E ReactNativeJS:     in Provider
04-11 16:51:04.706 11922 12001 E ReactNativeJS:     in MyProvider
04-11 16:51:04.706 11922 12001 E ReactNativeJS:     in _class2
04-11 16:51:04.706 11922 12001 E ReactNativeJS:     in RCTView
04-11 16:51:04.706 11922 12001 E ReactNativeJS:     in _class
04-11 16:51:04.706 11922 12001 E ReactNativeJS:     in RCTView
04-11 16:51:04.706 11922 12001 E ReactNativeJS:     in RootSiblingsWrapper
04-11 16:51:04.706 11922 12001 E ReactNativeJS:     in RCTView
04-11 16:51:04.706 11922 12001 E ReactNativeJS:     in AppContainer
```
This is what I see in the adb logcat. Later one I find the following

```
04-11 16:51:05.795 11922 11922 E MediaRecorder: stop failed: -1007
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: Exception in native call
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: java.lang.RuntimeException: stop failed.
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at android.media.MediaRecorder.stop(Native Method)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at com.google.android.cameraview.Camera1.stop(Camera1.java:145)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at com.google.android.cameraview.Camera1$1.onSurfaceDestroyed(Camera1.java:116)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at com.google.android.cameraview.PreviewImpl.dispatchSurfaceDestroyed(PreviewImpl.java:60)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at com.google.android.cameraview.TextureViewPreview$1.onSurfaceTextureDestroyed(TextureViewPreview.java:58)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at android.view.TextureView.releaseSurfaceTexture(TextureView.java:249)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at android.view.TextureView.onDetachedFromWindowInternal(TextureView.java:222)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at android.view.View.dispatchDetachedFromWindow(View.java:17598)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at android.view.ViewGroup.dispatchDetachedFromWindow(ViewGroup.java:3697)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at android.view.ViewGroup.dispatchDetachedFromWindow(ViewGroup.java:3697)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at android.view.ViewGroup.dispatchDetachedFromWindow(ViewGroup.java:3697)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at android.view.ViewGroup.removeViewInternal(ViewGroup.java:5249)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at android.view.ViewGroup.removeViewAt(ViewGroup.java:5196)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at com.facebook.react.uimanager.ViewGroupManager.removeViewAt(ViewGroupManager.java:80)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at com.facebook.react.uimanager.NativeViewHierarchyManager.manageChildren(NativeViewHierarchyManager.java:379)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at com.facebook.react.uimanager.UIViewOperationQueue$ManageChildrenOperation.execute(UIViewOperationQueue.java:180)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at com.facebook.react.uimanager.UIViewOperationQueue$1.run(UIViewOperationQueue.java:822)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:929)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at com.facebook.react.uimanager.UIViewOperationQueue.access$2100(UIViewOperationQueue.java:47)
04-11 16:51:05.799 11922 11922 E unknown:ReactNative: 	at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:989)
[... cut ...]
04-11 16:51:05.892 11922 11922 W System.err: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:482)
04-11 16:51:05.892 11922 11922 W System.err: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
```

And when it tries to start the camera I see many many of the following:
```
04-11 16:50:53.406   772 12446 E mm-camera: <SENSOR><ERROR> 497: port_sensor_handle_aec_update: flash_process  est_state= 0 ,state =0 ,torch_on =0, partial_flash_frame_skip =1, is_dual_cam = 0
```

While if the remote debugger is on, it all works without any issue.

sibelius, 2018-04-11:
what is this error about

`MediaRecorder: stop failed: -1007`?

can u reduce the size of the video and check if this works?

DevAlien, 2018-04-11:
I've set 740p already, same issue. still same thing, debugging works

DevAlien, 2018-04-18:
It makes me crazy that with the remote debugger on it works... and the fact that it fails on ios and android it is even crazier :S

jgfidelis, 2018-04-19:
https://stackoverflow.com/questions/15833694/mediarecorder-stop-stop-failed-1007

closed 2019-06-08 by stale[bot]