react-native-camera/react-native-camera #1459: RNCamera Preview Rotating Wrong Way

akriger, 2018-04-11:
### Which implementation are you using

*RNCamera* 

### Steps to reproduce
1. Create a view from the RNCamera Example
2. Open RNCamera view on iOS device and Rotate

### Expected behaviour
The camera preview should rotate Landscape left, Landscape right, upside down, portrait as the device rotates

### Actual behaviour
The camera preview rotates but it seems to rotate the wrong way and will not be correct unless you go back then reload the view again

### Environment
- **Node.js version**: 6.0.0
- **React Native version**: 0.53.3
- **React Native platform + platform version**: tested on physical iPad iOS 10.3.2

### react-native-camera
**Version**:  "master"

PavelPolyakov, 2018-08-11:
@akriger 
In my case if I have screen orientation fixed as `portrait` and I do rotate my phone to the `landscape` - camera preview rotates, as it want to stay in `portrait`. So the image in preview is `-90deg` rotated comparing to the one I see in real life.

Do you think this is the same issue? Or your's is different.

akriger, 2018-08-23:
@PavelPolyakov 
I'm not using expo and not specifying an orientation, but below is a video of what my issue is. I just updated everything and was doing some test. I found that when the device is rotated the view is initially wrong, but if I start to tilt the device down like I was going to lay it on a table it corrects itself. 

Still not right and not sure how to fix it.
[image]

akriger, 2018-08-23:
Didn't mean to close sorry

BigPun86, 2018-09-20:
i have the same issue. I have added the fixOrientation: true to the options when capturing an image. 

    "react-native": "0.53.3",
    "react-native-camera": "1.2.0"

@akriger any progress on this?

n1ru4l, 2018-12-12:
This was fixed https://github.com/react-native-community/react-native-camera/pull/1786.

If you still experience this issue with the newest version please reopen this issue.

closed 2018-12-12 by n1ru4l

t-hughes, 2019-01-10:
@akriger did you find a fix for your issue, I am having the same issue and updating to the latest version (1.8.0) still did not fix it. It only happens on iOS versions 10 or above for me.