react-native-camera/react-native-camera #1460: Conflicts when using pod Firebase/Database (react-native-firebase) on iOS

leonchabbey, 2018-04-12:
### Which implementation are you using

RNCamera

### Steps to reproduce
1. npm install react-native-firebase
2. Edit Podfile and add ```pod Firebase/Database``` and then ```pod update```
3. Try to build the app

### Expected behaviour
Should build without conflicting. It only happens with the specific ```pod Firebase/Database```.

### Actual behaviour
For reasons that I don't know, when I add Firebase/Database to my Podfile (I'm also using Firebase/Core and Firebase/Auth but they don't conflicts) it breaks RNCamera.
#PodFile:
```
# Uncomment the next line to define a global platform for your project
platform :ios, '9.0'

target 'app' do
  # Uncomment the next line if you're using Swift or would like to use dynamic frameworks
  # use_frameworks!

  # FCM
  pod 'Firebase/Messaging'
  
  # RNFirebase
  pod 'Firebase/Core'
  pod 'Firebase/Auth'
  pod 'Firebase/Database' # Causes the issue
end
```
[image]

[image]
[image]
[image]
[image]

### Environment
- **Node.js version**: v9.7.1
- **React Native version**: 55.1
- **React Native platform + platform version**: iOS 11.3, Android 8.0

### react-native-camera
**Version**: 1.0.3

leonchabbey, 2018-04-12:
Yep, I've found a way to make them work together.

My solution has been to not use Cocoapods for react-native-firebase and add the frameworks manually into my project.
1) Downloading all of them at: https://firebase.google.com/download/ios
2) Create a folder in my project under: ios/Firebase
3) Put all the frameworks I need in the folder and drag and drop them in xcode

(Note: I had to **not use the frameworks "GoogleToolboxForMac.framework" and "nanopb.framework"** because of duplication (Apple Match-O Linker dupication error...))

jgfidelis, 2018-04-12:
Cool... at our projects we avoid cocoapods at all, because it usually gives us too much problem with RN. Happy you found a solution and explained it for others, well done.

closed 2018-04-12 by jgfidelis