react-native-camera/react-native-camera #1461: Android error with googleVisionBarcodeType defaultProp

waitopiggu, 2018-04-13:
### Which implementation are you using

RNCamera

### Steps to reproduce
1. Initialize component without supplying prop `googleVisionBarcodeType`
2. Run on Android device
3. The component throws an error

### Expected behaviour
The component should mount using its default parameters

### Actual behaviour
RNCamera fails to initialize unless googleVisionBarcodeType prop is supplied as a number.

### Environment
- v9.11.1
- 0.55.2
- Android 6.0

### react-native-camera
master

>ExceptionsManager.js:71 Warning: Failed prop type: Invalid prop `googleVisionBarcodeType` of type `object` supplied to `Camera`, expected `number`.
>    in Camera (at Camera.js:7)
>    in Camera (at App.js:8)
>    in App (at renderApplication.js:33)
>    in RCTView (at View.js:60)
>    in View (at AppContainer.js:102)
>    in RCTView (at View.js:60)
>    in View (at AppContainer.js:122)
>    in AppContainer (at renderApplication.js:32)

[image]

sibelius, 2018-04-13:
The proptypes is wrong, can u send a pr to fix it?

waitopiggu, 2018-04-13:
@sibelius is `0` a safe value to set it as? The default prop is currently `{}`.

sibelius, 2018-04-13:
You should use exported constants instead of direct numbers

jgfidelis, 2018-04-13:
I think that 0 not woking is wrong...

Looking at the code in BarcodeDetectorUtils:
```
public class BarcodeFormatUtils {

  public static final SparseArray<String> FORMATS;
  public static final Map<String, Integer> REVERSE_FORMATS;

  private static final String UNKNOWN_FORMAT_STRING = "UNKNOWN_FORMAT";
  private static final int UNKNOWN_FORMAT_INT = -1;

  static {
    // Initialize integer to string map
    SparseArray<String> map = new SparseArray<>();
    map.put(Barcode.CODE_128, "CODE_128");
    map.put(Barcode.CODE_39, "CODE_39");
    map.put(Barcode.CODE_93, "CODE_93");
    map.put(Barcode.CODABAR, "CODABAR");
    map.put(Barcode.DATA_MATRIX, "DATA_MATRIX");
    map.put(Barcode.EAN_13, "EAN_13");
    map.put(Barcode.EAN_8, "EAN_8");
    map.put(Barcode.ITF, "ITF");
    map.put(Barcode.QR_CODE, "QR_CODE");
    map.put(Barcode.UPC_A, "UPC_A");
    map.put(Barcode.UPC_E, "UPC_E");
    map.put(Barcode.PDF417, "PDF417");
    map.put(Barcode.AZTEC, "AZTEC");
    FORMATS = map;

    // Initialize string to integer map
    Map<String, Integer> rmap = new HashMap<>();
    for (int i = 0; i < map.size(); i++) {
      rmap.put(map.valueAt(i), map.keyAt(i));
    }

    rmap.put("ALL", 0);
    REVERSE_FORMATS = Collections.unmodifiableMap(rmap);
  }

  public static String get(int format) {
    return FORMATS.get(format, UNKNOWN_FORMAT_STRING);
  }

  public static int get(String format) {
    if (REVERSE_FORMATS.containsKey(format)) {
      return REVERSE_FORMATS.get(format);
    }

    return UNKNOWN_FORMAT_INT;
  }
}
```
`rmap.put("ALL", 0);` so 0 should be read all?
A good fix would be to create a `none` value in the constants and make:
`googleVisionBarcodeType: (CameraManager.GoogleVisionBarcodeDetection || {}).BarcodeType.None`
in defaultProps. If no one fix this, I will try during the weekend.

closed 2018-04-13 by sibelius