react-native-camera/react-native-camera #1466: onFacesDetect loop after change Scene

raulcrash, 2018-04-13:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. Copy example [FaceDetector ](https://github.com/react-native-community/rncamera-example/blob/master/App.js).
2. Modify line 106:
` onFacesDetected = ({ faces }) => this.setState({ faces });`
**To**
```
  onFacesDetected = ({ faces }) => {
        this.setState({ faces });
        if(faces.length>0){
          if(faces[0].leftEyeOpenProbability<0.7 && faces[0].rightEyeOpenProbability<0.7) {
            console.warn("","Closed!");
          }else {
            console.log("","Open!");
          }
        }
      }
```
3. Create a button to change scene:
```
import { Actions } from 'react-native-router-flux';
........
goTestGPS() {
   Actions.testeGPS();
}
....
render() {
      <RNCamera 
      .......
             <TouchableOpacity style={styles.flipButton} onPress={this.goTestGPS.bind(this)}> 
                     <Text style={styles.flipText}> BACK </Text>
             </TouchableOpacity>
       ......
       </RNCamera>
}
```
4. Mode Debug App, after redirect Scene onFacesDetect is call everytime.

### Expected behaviour
After changing the scene, the camera should not listen onFacesDetected.

### Actual behaviour
Even when change scene, the scene before call onFacesDetected everytime

### Environment
- **Node.js version**: 5.8.0
- **React Native version**: 0.46.4
- **React Native platform + platform version**: tested on Android 7.0

### react-native-camera
**Version**: "master"

sibelius, 2018-04-13:
what is change scene?

moving to another route?

this is happening because the screen is will mounted in the stack

you should check if your component `isFocused` and stop rendering camera and stop listening for onFacesDetected event

jgfidelis, 2018-04-13:
with react-navigation, you could do:

```
<RNCamera
onFacesDetected={this.props.navigation.isFocused() ? this.onFacesDetected : null}
</RNCamera>
```

raulcrash, 2018-04-13:
Thank @jgfidelis, its work!! Nice

phanvinh876, 2020-07-18:
Iam using @jgfidelis way but still facing this issue

closed 2020-09-26 by stale[bot]