react-native-camera/react-native-camera #1467: Error while updating property 'barcodeScannerEnabled' of view managed by : RNCamera

Moussawi7, 2018-04-14:
### Which implementation are you using
RNCamera

### Steps to reproduce
1.install the plugin : npm install --save https://github.com/react-native-community/react-native-camera
2.link it :react-native-link react-native-camera
3.add the js code

```
<RNCamera
          ref={(cam) => {
            this.camera = cam;
          }}
          ratio="1:1"
          permissionDialogTitle={'Permission to use camera'}
          permissionDialogMessage={'We need your permission to use your camera phone'}
          onBarCodeRead={(e)=>this.onBarCodeRead(e)}
          style={Styles.camera}/>
```

### Expected behaviour
Scan QR code

### Actual behaviour
throw an error saying "Error while updating property 'barcodeScannerEnabled' of view managed by : RNCamera"
[image]

### Environment
- **Node.js version**:8.0.0
- **React Native version**:0.53.3
- **React Native platform + platform version**: Android 6.0.1

### react-native-camera
**Version**: "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

wassgha, 2018-04-14:
Having this bug too

sibelius, 2018-04-14:
it could be related to this commit https://github.com/react-native-community/react-native-camera/pull/1464

can u try with a commit before that one?

jgfidelis, 2018-04-14:
Does it happen as soon as the app starts?

It doesn't happen on my emulator, it happens only on when you point it to a barcode?

jgfidelis, 2018-04-14:
Can you guys tell me if #1469 fixes the bug?

closed 2018-04-14 by wassgha

ajaykumar97, 2018-11-26:
Following configuration worked for me: 
`implementation (project(':react-native-camera')) {
        implementation ("com.google.android.gms:play-services-vision:10.0.1") {
            force = true
        }

        exclude group: "com.google.android.gms"
        implementation 'com.android.support:exifinterface:27.+'
    }
implementation 'com.google.android.gms:play-services-base:10.0.1'
`

Make sure that all google-play-services implementations have same version(10.0.1 in my case).

AndreyPatseiko, 2018-11-28:
Please update the documentation as the starting example uses **onGoogleVisionBarcodesDetected** function for read code. (in BadInstagramCloneApp)

n1ru4l, 2018-11-30:
@AndreyPatseiko Feel free to send a pull request that improves the documentation if there are any misunderstandings.

PR #1469 (2018-04-14): fix(barcode): better name google variables and correct init
try to fix #1467 

Better name the google barcode detector variables, they had very similar names with the old barcode detector which led to confusion.

Initialize the old barcode detector lazily