react-native-camera/react-native-camera #1485: How to preview picture after take a picture

mutiadianita, 2018-04-16:
I'm using RNCamera and I want to preview the picture I took from the camera. Is it possible to do it in RNCamera? Or can you give me an example to do that? I'm still newbie

jgfidelis, 2018-04-16:
After you takePictureAsync() resolves, you can pass the image uri to `ImageBackground` via state.

Like this:
```
takePicture = () => {
  const { uri } = await this.camera.takePictureAsync();
  this.setState({ imageUri: uri });
}

render() {
  const { imageUri } = this.state;
  if (imageUri) {
    return <ImageBackground source={imageUri}/>;
  }
  return ( 
    <RNCamera... 
   );
}
```

closed 2018-04-16 by jgfidelis

sibelius, 2018-04-16:
Recipe

mutiadianita, 2018-04-17:
Thank youuuu!

swingywc, 2018-10-30:
Thank you jgfidelis. Your solution save my days :)

IbrahimSheikh24, 2019-11-18:
Hi @jgfidelis ,
ImageUri is not binding to ImageView after capturing. I have given the permissions as below in manifest
<uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />

My code is as below
----------------------------
import React from 'react';
import { View, StyleSheet, Text, TouchableOpacity, Image,  } from 'react-native'
import { RNCamera } from 'react-native-camera';

export default class Camera extends React.Component {

  constructor(props) {
    super(props);
    this.state = {
      path: null,
      data: null
    }
  }

  renderCamera() {
    return (
      <RNCamera
        ref={ref => {
          this.camera = ref;
        }}
        style={styles.preview}
        type={RNCamera.Constants.Type.back}
        flashMode={RNCamera.Constants.FlashMode.on}
        captureAudio={false}
        androidCameraPermissionOptions={{
          title: 'Permission to use camera',
          message: 'We need your permission to use your camera',
          buttonPositive: 'Ok',
          buttonNegative: 'Cancel',
        }}
      >

        <View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
          <TouchableOpacity
            onPress={() => {
              this.takePicture()
            }}
            style={styles.capture}>
            <Text style={{ fontSize: 14 }}> SNAP </Text>
          </TouchableOpacity>
        </View>
      </RNCamera>
    )
  }
  
  renderImage() {
    return (
      <View>
        <Image
          source={{ uri: this.state.path }}
          style={styles.preview}
        />
        <Text
          style={styles.cancel}
          onPress={() => this.setState({ path: null })}
        >Cancel
        </Text>
      </View>
    );
  }

  render() {
    return (
      <View style={styles.container}>
        {this.state.path ? this.renderImage() : this.renderCamera()}
      </View>
    );
  }

  takePicture = async () => {
    try {
      const options = { quality: 0.5, pauseAfterCapture:true };
      const data = await this.camera.takePictureAsync(options);
        this.setState({ path: data.uri, data : data });
        console.log('Path to image: ' + data.uri);
      CameraRoll.saveToCameraRoll(data.uri,"photo")
    } catch (err) {
      console.log('err: ', err);
    }
  };
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    flexDirection: 'column',
    backgroundColor: 'blue',
  },
  preview: {
    flex: 1,
    justifyContent: 'flex-end',
    alignItems: 'center',
  },
  capture: {
    flex: 0,
    backgroundColor: '#fff',
    borderRadius: 5,
    padding: 15,
    paddingHorizontal: 20,
    alignSelf: 'center',
    margin: 20,
  },
  cancel: {
    position: 'absolute',
    right: 20,
    top: 20,
    backgroundColor: 'transparent',
    color: '#FFF',
    fontWeight: '600',
    fontSize: 17,
  },
});

Please help me on this, where I am mistaking. Thanks