react-native-camera/react-native-camera #1486: java.lang.ArrayIndexOutOfBoundsException when loading camera view

nvme978, 2018-04-16:
Using RNCamera 
react-native-camera version 1.1.1
"react": "16.0.0",
"react-native": "0.51.0",

Testing on Genymotion

### Actual behaviour
App crashes with the following logcat on Android
`04-16 06:51:36.372   376  2146 V EmulatedCamera_CallbackNotifier: **** Currently enabled messages:
04-16 06:51:36.372   376  2146 V EmulatedCamera_CallbackNotifier:     CAMERA_MSG_ERROR
04-16 06:51:36.372   376  2146 V EmulatedCamera_CallbackNotifier:     CAMERA_MSG_FOCUS
04-16 06:51:36.372   376  2146 V EmulatedCamera_CallbackNotifier:     CAMERA_MSG_ZOOM
04-16 06:51:36.372   376  2146 V EmulatedCamera_CallbackNotifier:     CAMERA_MSG_PREVIEW_METADATA
04-16 06:51:36.372  2007  2007 E AndroidRuntime: FATAL EXCEPTION: main
04-16 06:51:36.372  2007  2007 E AndroidRuntime: Process: com.aerobittech.okfit, PID: 2007
04-16 06:51:36.372  2007  2007 E AndroidRuntime: java.lang.ArrayIndexOutOfBoundsException: length=115200; index=115679
04-16 06:51:36.372  2007  2007 E AndroidRuntime: 	at org.reactnative.camera.RNCameraView$1.rotateImage(RNCameraView.java:111)
04-16 06:51:36.372  2007  2007 E AndroidRuntime: 	at org.reactnative.camera.RNCameraView$1.onFramePreview(RNCameraView.java:126)
04-16 06:51:36.372  2007  2007 E AndroidRuntime: 	at com.google.android.cameraview.CameraView$CallbackBridge.onFramePreview(CameraView.java:558)
04-16 06:51:36.372  2007  2007 E AndroidRuntime: 	at com.google.android.cameraview.Camera1.onPreviewFrame(Camera1.java:713)
04-16 06:51:36.372  2007  2007 E AndroidRuntime: 	at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1110)
04-16 06:51:36.372  2007  2007 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:102)
04-16 06:51:36.372  2007  2007 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:148)
04-16 06:51:36.372  2007  2007 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:5417)
04-16 06:51:36.372  2007  2007 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
04-16 06:51:36.372  2007  2007 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
04-16 06:51:36.372  2007  2007 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
04-16 06:51:36.374   682  1039 W ActivityManager:   Force finishing activity com.aerobittech.okfit/.MainActivity`

kanalasumant, 2018-04-18:
Hey @nvme978 can't tell if this is the error due to which the Camera component doesn't render and force closes app on my phone, but this seems to start a few days ago. The camera was being rendered perfectly. No code had been changed. It looked like when I restarted the project a few days back some new packages might have been downloaded from the gradle build. Not sure if that's causing the issue or not. Hoping to hear if you resolved this. Thank you!

ramzed, 2018-04-18:
Hello,
I have the same issue. This error appears in 50% of starts of my application. I can't understand why it happens...

Using RNCamera
react-native-camera version 1.1.1
"react": "16.3.0-alpha.1",
"react-native": "0.53.3",

ayobami303, 2018-04-19:
am also having similar error. app crashes when its minimized and opened again.
a fix would be appreciated

jgfidelis, 2018-04-20:
Is the same error happening on the actual device? Is there any log? The log on the thread looks like a Emulator one.

dieselsaurav, 2018-04-26:
04-25 23:07:03.590 14222-14222/? E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.zopper.pos.android, PID: 14222
    java.lang.RuntimeException: com.facebook.react.uimanager.IllegalViewOperationException: Trying to remove a view index above child count 0 view tag: 11
     detail: View tag:-1
      children(0): [
     ],
      indicesToRemove(1): [
    0,
     ],
      tagsToDelete(1): [
    395,
     ]
    
        at com.facebook.react.bridge.ReactContext.handleException(ReactContext.java:311)
        at com.facebook.react.bridge.GuardedRunnable.run(GuardedRunnable.java:23)
        at android.os.Handler.handleCallback(Handler.java:822)
        at android.os.Handler.dispatchMessage(Handler.java:104)
        at android.os.Looper.loop(Looper.java:207)
        at android.app.ActivityThread.main(ActivityThread.java:5811)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:791)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:681)
     Caused by: com.facebook.react.uimanager.IllegalViewOperationException: Trying to remove a view index above child count 0 view tag: 11
     detail: View tag:-1
      children(0): [
     ],
      indicesToRemove(1): [
    0,
     ],
      tagsToDelete(1): [
    395,
     ]
    
        at com.facebook.react.uimanager.NativeViewHierarchyManager.manageChildren(NativeViewHierarchyManager.java:351)
        at com.facebook.react.uimanager.UIViewOperationQueue$ManageChildrenOperation.execute(UIViewOperationQueue.java:179)
        at com.facebook.react.uimanager.UIViewOperationQueue$1.run(UIViewOperationQueue.java:821)
        at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:928)
        at com.facebook.react.uimanager.UIViewOperationQueue.access$2100(UIViewOperationQueue.java:46)
        at com.facebook.react.uimanager.UIViewOperationQueue$2.runGuarded(UIViewOperationQueue.java:886)
        at com.facebook.react.bridge.GuardedRunnable.run(GuardedRunnable.java:21)
        	... 7 more

04-25 23:11:53.828 14683-14683/? E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.zopper.pos.android, PID: 14683
    java.lang.ArrayIndexOutOfBoundsException: length=115200; index=115919
        at org.reactnative.camera.RNCameraView$1.rotateImage(RNCameraView.java:111)
        at org.reactnative.camera.RNCameraView$1.onFramePreview(RNCameraView.java:126)
        at com.google.android.cameraview.CameraView$CallbackBridge.onFramePreview(CameraView.java:558)
        at com.google.android.cameraview.Camera1.onPreviewFrame(Camera1.java:713)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1290)
        at android.os.Handler.dispatchMessage(Handler.java:111)
        at android.os.Looper.loop(Looper.java:207)
        at android.app.ActivityThread.main(ActivityThread.java:5811)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:791)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:681)

dieselsaurav, 2018-04-26:
this is what I got from the android crash report.

java.lang.ArrayIndexOutOfBoundsException: 
  at org.reactnative.camera.RNCameraView$1.rotateImage (RNCameraView.java:111)
  at org.reactnative.camera.RNCameraView$1.onFramePreview (RNCameraView.java:126)
  at com.google.android.cameraview.CameraView$CallbackBridge.onFramePreview (CameraView.java:558)
  at com.google.android.cameraview.Camera1.onPreviewFrame (Camera1.java:713)
  at android.hardware.Camera$EventHandler.handleMessage (Camera.java:1290)
  at android.os.Handler.dispatchMessage (Handler.java:111)
  at android.os.Looper.loop (Looper.java:207)
  at android.app.ActivityThread.main (ActivityThread.java:5811)
  at java.lang.reflect.Method.invoke (Native Method)
  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:791)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:681)

jpudysz, 2018-04-26:
Same here, reproduced on Galaxy Grand Prime
Crash report from appcenter:

```
org.reactnative.camera.RNCameraView$1.rotateImage
RNCameraView.java - line 111

java.lang.ArrayIndexOutOfBoundsException: length=115200; index=116159

1
org.reactnative.camera.RNCameraView$1.rotateImage RNCameraView.java:111
2
org.reactnative.camera.RNCameraView$1.onFramePreview RNCameraView.java:126
3
com.google.android.cameraview.CameraView$CallbackBridge.onFramePreview CameraView.java:558
4
com.google.android.cameraview.Camera1.onPreviewFrame Camera1.java:713
5
android.hardware.Camera$EventHandler.handleMessage Camera.java:1160
6
android.os.Handler.dispatchMessage Handler.java:102
7
android.os.Looper.loop Looper.java:135
8
android.app.ActivityThread.main ActivityThread.java:5910
9
java.lang.reflect.Method.invoke Method.java
10
java.lang.reflect.Method.invoke Method.java:372
11
com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run ZygoteInit.java:1405
12
com.android.internal.os.ZygoteInit.main ZygoteInit.java:1200
```

jpudysz, 2018-04-30:
Exactly same stack trace & issue with Galaxy S7 Edge

nijolas, 2018-05-01:
Had a cursory look into what is going on, at least on my Galaxy S7. `onFramePreview` is normally being called at 960x720. When the exception occurs it's being called at 1440x1080.

I don't know enough to say anything more that that, just thought that might help someone diagnose it (this is on master, btw).

Locals:
```
this = {RNCameraView$1@6481} 
imageData = {byte[1036800]@6486} 
height = 1440
width = 1080
ex = {ArrayIndexOutOfBoundsException@6487} "java.lang.ArrayIndexOutOfBoundsException: length=1036800; index=1037879"
```

mehdinejati, 2018-05-10:
"react-native-camera": "^1.1.1",
"react-native": "0.55.2",

android version 4.2.2
device HUAWEI G730-U10

Fatal Exception: java.lang.ArrayIndexOutOfBoundsExceptionlength=115200; index=115799 Raw Text
--
  | org.reactnative.camera.RNCameraView$1.rotateImage (RNCameraView.java:111)
  | org.reactnative.camera.RNCameraView$1.onFramePreview (RNCameraView.java:126)

closed 2018-09-25 by SimonErm