react-native-camera/react-native-camera #1488: App force closes upon camera component render

kanalasumant, 2018-04-16:
This issue was already solved for me, but it seems to appear again. I've modified the compile dependencies in app/build.gradle to ***com.google.android.gms:play-services-vision:12.0.1** recently too.
I've received an **OTA** update by the way but the version of android stayed **7.0.**

Below is the camera component. Do I need any special permission upon app load for the app to render.
```
import React, { Component } from "react";
import { View, Text, TouchableOpacity, StyleSheet } from "react-native";

import { RNCamera } from "react-native-camera";
import RNFS from "react-native-fs";

import ActionButton from "./action-btn";

import { path } from "./../assets/data/config";

getImageNameFromPath = imagePathAsString => {
  let imagePathAsArray = imagePathAsString.split('/');
  return imagePathAsArray[imagePathAsArray.length - 1];
}

moveImageFromCacheToProjectFolder = async ({ project_name }, imagePathAtCache) => {
  const imageName = getImageNameFromPath(imagePathAtCache);
  return await RNFS.moveFile(imagePathAtCache, `${path}AppMyProject/${project_name}/${imageName}`);
}

export default class Camera extends Component {
  state = {
    flash: 0
  }

  backToGallery = () => this.props.goToGalleryView();

  takePicture = async function () {
    try {
      if (this.camera) {
        const options = { quality: 0.5, base64: true };
        const data = await this.camera.takePictureAsync(options);
        await moveImageFromCacheToProjectFolder(this.props, data.uri);
        alert('Image saved');
      }
    } catch (err) {
      alert(`Image Not saved due to: ${err}`);
    }
  };

  toggleFlash = () => this.setState(prevState => ({
    flash: (prevState.flash === 0) ? 1 : 0
  }));

  render() {
    return (
      <View style={styles.container}>
        <RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          style={styles.preview}
          type={RNCamera.Constants.Type.back}
          flashMode={this.state.flash}
          permissionDialogTitle={"Permission to use camera"}
          permissionDialogMessage={
            "We need your permission to use your camera phone"
          }
        />
        <View style={{ flex: 0, flexDirection: "row" }}>
          <View style={{ flex: 1 }}>
            <ActionButton goToView={this.backToGallery} icon_name="times" />
          </View>
          <View style={{ flex: 1, justifyContent: "center" }}>
            <TouchableOpacity
              onPress={this.takePicture.bind(this)}
              style={styles.capture}
            >
              <Text style={{ fontSize: 14 }}>SNAP</Text>
            </TouchableOpacity>
          </View>
          <View style={{ flex: 1 }}>
            <ActionButton goToView={this.toggleFlash} icon_name={this.state.flash ? "flash" : "eye-slash"} />
          </View>
        </View>
      </View>
    )
  }
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    flexDirection: "column",
    backgroundColor: "black"
  },
  preview: {
    flex: 1,
    justifyContent: "flex-end",
    alignItems: "center"
  },
  capture: {
    flex: 0,
    backgroundColor: "#fff",
    borderRadius: 5,
    padding: 15,
    paddingHorizontal: 20,
    alignSelf: "center",
    margin: 20
  }
});
```

### Does it work with Expo Camera?
RNCamera is based on Expo Camera, if this does not work with Expo Camera it won't work with RNCamera as well.
You should open an issue there as well, so we can cooperate in a solution.

### Expected behaviour
Camera component should render.

### Actual behaviour
While trying to render the camera component app crashes/force closes.

### Environment
- **Node.js version**: 8.9.1
- **React Native version**: 0.55.2
- **React Native platform + platform version**: Android 7.0

### react-native-camera
**Version**: 1.0.3

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgfidelis, 2018-04-16:
What is the error? Probably something related to gms, how is it in your build.gradle?

jgfidelis, 2018-04-16:
These errors are usually (a) permission related or (b) wrong build.gradle configuration

kanalasumant, 2018-04-16:
I have enabled camera permission in androidmanifest.xml and here are is my app/build.gradle config:
```
apply plugin: "com.android.application"

import com.android.build.OutputFile

project.ext.react = [
    entryFile: "index.js"
]

apply from: "../../node_modules/react-native/react.gradle"

def enableSeparateBuildPerCPUArchitecture = false

/**
 * Run Proguard to shrink the Java bytecode in release builds.
 */
def enableProguardInReleaseBuilds = false

android {
    compileSdkVersion 23
    buildToolsVersion "23.0.3"

    defaultConfig {
        applicationId "com.appmyproject"
        minSdkVersion 16
        targetSdkVersion 22
        versionCode 1
        versionName "1.0"
        ndk {
            abiFilters "armeabi-v7a", "x86"
        }
    }
    splits {
        abi {
            reset()
            enable enableSeparateBuildPerCPUArchitecture
            universalApk false  // If true, also generate a universal APK
            include "armeabi-v7a", "x86"
        }
    }
    signingConfigs {
        release {
            if (project.hasProperty('MYAPP_RELEASE_STORE_FILE')) {
                storeFile file(MYAPP_RELEASE_STORE_FILE)
                storePassword MYAPP_RELEASE_STORE_PASSWORD
                keyAlias MYAPP_RELEASE_KEY_ALIAS
                keyPassword MYAPP_RELEASE_KEY_PASSWORD
            }
        }
    }
    buildTypes {
        release {
            minifyEnabled enableProguardInReleaseBuilds
            proguardFiles getDefaultProguardFile("proguard-android.txt"), "proguard-rules.pro"
            signingConfig signingConfigs.release
        }
    }
    // applicationVariants are e.g. debug, release
    applicationVariants.all { variant ->
        variant.outputs.each { output ->
            // For each separate APK per architecture, set a unique version code as described here:
            // http://tools.android.com/tech-docs/new-build-system/user-guide/apk-splits
            def versionCodes = ["armeabi-v7a":1, "x86":2]
            def abi = output.getFilter(OutputFile.ABI)
            if (abi != null) {  // null for the universal-debug, universal-release variants
                output.versionCodeOverride =
                        versionCodes.get(abi) * 1048576 + defaultConfig.versionCode
            }
        }
    }
}

dependencies {
    compile project(':react-native-vector-icons')
    compile project(':react-native-static-server')
    compile project(':react-native-fs')
    compile project(':react-native-push-notification')
    compile (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
        compile 'com.android.support:exifinterface:25.+'
        compile ('com.google.android.gms:play-services-vision:12.0.1') {
            force = true
        }
    }
    compile fileTree(dir: "libs", include: ["*.jar"])
    compile "com.android.support:appcompat-v7:23.0.1"
    compile "com.facebook.react:react-native:+"  // From node_modules
}

// Run this once to be able to run the application with BUCK
// puts all compile dependencies into folder libs for BUCK to use
task copyDownloadableDepsToLibs(type: Copy) {
    from configurations.compile
    into 'libs'
}
```

Here is android/build.gradle config too:
```
// Top-level build file where you can add configuration options common to all sub-projects/modules.

buildscript {
    repositories {
        jcenter()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:2.2.3'

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        maven { url "https://jitpack.io" }
        maven { url "https://maven.google.com" }
        mavenLocal()
        jcenter()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
    }
}
```
Let me know if this helps.
**I'm assuming the only thing changed is my react-native version from 0.52.2 to 0.55.2. I'm assuming there's a bug with it. I can go back to the old version by cloning my whole repo.

Thanks for quickly getting back.

jgfidelis, 2018-04-16:
Here is the relevant part of our build.gradle:
```
implementation (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
        exclude group: "com.android.support"
    }
implementation "com.android.support:exifinterface:27.0.2"
```

The best thing would be to get the stacktrace of the error that is happening. Can you run it with Android Studios open and see the error?

Maybe this in yours can help:
```
compile (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
        exclude group: "com.android.support"
    }
compile 'com.android.support:exifinterface:25.+'
compile ('com.google.android.gms:play-services-vision:12.0.1') {
    force = true
}
```

kanalasumant, 2018-04-16:
Yes. I'm constantly reminded to check for errors in android studio since some errors get swallowed? I'll start with the latter one and if it doesn't work will check with android studio. By the way just wanted to know have any others or your team of contributors ran this package with the latest 0.55.2 RN. Any issues encountered. Thanks!

jgfidelis, 2018-04-16:
We still did not try with RN 0.55.2, our projects are using 0.53.

kanalasumant, 2018-04-17:
Also does it make a difference if the Android SDK Platform version is 27. I downloaded the recommended 23 from Studio. Could anyone give insight onto how my RN project would know about this and use that lower version. It might be the one causing an issue, although the only way to know is to test it.

jgfidelis, 2018-04-17:
Android is complicated, too many variables. React Native 0.54 has React 16.3 with the Context API.

Without the stack trace or an error message is difficult to help more.

kanalasumant, 2018-04-17:
I'm currently re installing android studio from scratch. Will see if that would help

sibelius, 2018-04-17:
please get native logs as well

kanalasumant, 2018-04-17:
The logcat is bombarded with phone's logs. The error log gets buried and other logs take over. This is when I connect my phone through adb

kanalasumant, 2018-04-17:
So from the previous version when the Camera component wasn't crashing, some other react-native code changed, but unsure if that could effect it or not. So previously the camera component was directly rendered under a Tab in TabNavigator **(react-navigation)**, but it took the whole view of the screen. but now it's directly rendered under a DrawerNavigator. Not sure if that would make a difference or not.

kanalasumant, 2018-04-17:
It's definitely my system. Re installed Android Studio. Still no luck. Going back to the other system where it works for now.

kanalasumant, 2018-04-17:
Oh boy now the version which works in the older system, doesn't seem to work. :( It might be that some packages were downloaded additionally when I ran react-native run-android now. Not sure which packages were updated. It was working just a few days ago. I didn't change a single line of code.

closed 2018-04-17 by sibelius