react-native-camera/react-native-camera #1490: Could not find method compileOnly()

rockcs1992, 2018-04-17:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. init a project with react-native-cli
2. follow the docs on this page 
https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md 
3. run `react-native run-android`

### Expected behaviour
The build should succeed
### Actual behaviour
The build failed with the following error: 
`Could not find method compileOnly() for arguments [com.facebook.react:react-native:+] on object of type org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler`

### Environment
- **Node.js version**: 8.9.4
- **React Native version**: 0.55.2
- **React Native platform + platform version**: Android 8.1

### react-native-camera
**Version**: 1.1.1

sibelius, 2018-04-17:
Use gradle 3

closed 2018-04-17 by sibelius

dgurns, 2018-04-17:
This worked for me - thanks @rockcs1992

waitopiggu, 2018-04-17:
@ldstein Had to change `android/app/build.gradle` `buildToolsVersion "23.0.1"` to `buildToolsVersion "26.0.1"` too. Many thanks.

rockcs1992, 2018-04-18:
@ldstein It works. Thank you so much

jjzazuet, 2018-04-21:
It's unfortunate that usage of this component now requires upgrading to Gradle 3. Doing so deviates us from the standard React Native project template configuration which is, if I'm not mistaken, still tied to Gradle 2.14.

```
distributionUrl=https\://services.gradle.org/distributions/gradle-2.14.1-all.zip
```

I'm guessing this must be an upstream convention the Facebook folks have to remain compatible with Android Studio or something. I'm wondering why their RN template project's Gradle wrapper version has not been upgraded for such a long time.

Cheers!

jsina, 2018-04-21:
@ldstein I've had the problem yet. could you please copy your change file here?

tinybyt, 2018-04-22:
worked for me too!
thanks a bunch @ldstein

jgfidelis, 2018-04-22:
https://itnext.io/install-react-native-maps-with-gradle-3-on-android-44f91a70a395

Article about the rn-maps library that may help people here, just need to adapt to rn-camera

ldstein, 2018-04-24:
I would also like to understand this.  This was the first package I hit which didn't install and work with `create-react-native-app`, leading rather quickly to a deep dive into Android toolchain fundamentals. 

I'm new to React Native and started by following the tutorial at https://facebook.github.io/react-native/docs/getting-started.html. Nothing in that guide mentions handling Gradle requirements when installing React-Native components.

Was well worth the time and I'm all the more wiser, but still unclear why react-native-camera requires Gradle 3 in order to build.

HAPENLY, 2018-04-24:
hi ,@ldstein I use your way, solve encountered this error, has anyone met?

`Could not resolve all files for configuration ':classpath'.
   > Could not find com.android.tools.build:gradle:3.1.0.
     Searched in the following locations:
         https://jcenter.bintray.com/com/android/tools/build/gradle/3.0.0/gradle-3.1.0.pom
         https://jcenter.bintray.com/com/android/tools/build/gradle/3.0.0/gradle-3.1.0.jar
     Required by:
         project :`

KarimFereidooni, 2018-04-30:
You can fix this by navigating to node_modules/react-native-maps/lib/android/build.gradle and changing these lines

    compileOnly "com.facebook.react:react-native:+"
    implementation "com.google.android.gms:play-services-base:$googlePlayServicesVersion"
    implementation "com.google.android.gms:play-services-maps:$googlePlayServicesVersion"
    implementation "com.google.maps.android:android-maps-utils:$androidMapsUtilsVersion"

to

    provided "com.facebook.react:react-native:+"
    compile "com.google.android.gms:play-services-base:$googlePlayServicesVersion"
    compile "com.google.android.gms:play-services-maps:$googlePlayServicesVersion"
    compile "com.google.maps.android:android-maps-utils:$androidMapsUtilsVersion"

GetSource1234, 2018-05-14:
Still, have this issue even after dirty workarounds.

zyfccc, 2018-05-17:
Spent a whole day on it. Finally, I have this which is working:

in android/gradle.properties: 

```
android.useDeprecatedNdk=true
android.enableAapt2=false
```

in android/build.gradle:

```
uildscript {
    repositories {
        jcenter()
        maven {
            url 'https://maven.google.com'
            name 'Google'
        }
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:3.1.0'

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        mavenLocal()
        jcenter()
        maven {
            url 'https://maven.google.com'
            name 'Google'
        }
        maven { url "https://jitpack.io" }
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        configurations.all {
            resolutionStrategy {
                force 'com.facebook.android:facebook-android-sdk:4.28.0'
            }
        }
    }
}

subprojects {
    project.configurations.all {
        resolutionStrategy.eachDependency { details ->
            if (details.requested.group == 'com.android.support'
              && !details.requested.name.contains('multidex') ) {
                details.useVersion "26.0.1"
            }
        }
    }
}
```

in android/app/build.gradle:

```
android {
    compileSdkVersion 26
    buildToolsVersion "26.0.1"

    defaultConfig {
        applicationId "appName"
        minSdkVersion 16
        targetSdkVersion 22
        versionCode 1
        versionName "1.0"
        ndk {
            abiFilters "armeabi-v7a", "x86"
        }
    }

...

dependencies {
    implementation project(':react-native-camera')
    ...
    implementation fileTree(dir: "libs", include: ["*.jar"])
    implementation 'com.android.support:appcompat-v7:26.0.1'
    implementation "com.facebook.react:react-native:+"  // From node_modules
}
```

in android/gradle/gradle-wrapper.properties:

```
distributionBase=GRADLE_USER_HOME
distributionPath=wrapper/dists
zipStoreBase=GRADLE_USER_HOME
zipStorePath=wrapper/dists
distributionUrl=https\://services.gradle.org/distributions/gradle-4.4-all.zip
```

Hope it helps...

sibelius, 2018-05-17:
@zyfccc would u mind sending this to our docs?

maybe another .md file inside docs folder

zyfccc, 2018-05-17:
@sibelius Yes my pleasure. Please see the created pull request.

ios-dev-newbie, 2018-05-18:
I have tried the changes mentioned by @ldstein  but not working for me.

I am getting below error:
* What went wrong:
Execution failed for task ':react-native-camera:generateDebugRFile'.
> Unknown android attribute 'CameraView'

my package.json is as follows:
dependencies": {
    "react": "16.3.1",
    "react-native": "0.53.3",
    "react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera",
  }

Please let me know if I am missing something.

AliZeynalov, 2018-06-07:
@ldstein Man you saved my life...

bnolan, 2018-06-22:
Edit `node_modules/react-native-camera/android/build.gradle`:

Change:

```
  compileOnly 'com.facebook.react:react-native:+'
  implementation "com.google.zxing:core:3.2.1"
  implementation "com.drewnoakes:metadata-extractor:2.9.1"
  implementation "com.google.android.gms:play-services-vision:$googlePlayServicesVersion"
  implementation "com.android.support:exifinterface:$supportLibVersion"

  implementation 'com.github.react-native-community:cameraview:eb9f726215bbbfcc03a54db92e480d4d02e0c001'
```

to

```
  provided 'com.facebook.react:react-native:+'
  compile "com.google.zxing:core:3.2.1"
  compile "com.drewnoakes:metadata-extractor:2.9.1"
  compile "com.google.android.gms:play-services-vision:$googlePlayServicesVersion"
  compile "com.android.support:exifinterface:$supportLibVersion"

  compile 'com.github.react-native-community:cameraview:eb9f726215bbbfcc03a54db92e480d4d02e0c001'
```

You have to re-do it after each time you `npm install` .

umar2731, 2018-07-28:
Hello everyone, i am using react native 0.56 version right now, and i get some error with the `build.gradle` on react-native-share mudule inside node module, and i just ccomment the dependencies code like this

```dependencies {
    provided "com.facebook.react:react-native:+"
    // compileOnly "com.facebook.react:react-native:${safeExtGet('reactNativeVersion', '+')}"
    // compile "com.android.support:appcompat-v7:${safeExtGet('supportLibVersion', '27.1.0')}"
}```

akhtarvahid, 2018-08-01:
### Problem that i am facing while running the application

**A problem occurred evaluating project ':react-native-share'.
> Could not find method compileOnly() for arguments [com.facebook.react:react-native:+] on object of type org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler**

Can anybody suggest me solution please...

banafsh13, 2018-08-13:
@ldstein thank you, works for me!

jose2007kj, 2018-10-01:
> provided

solved my issue in leniar-grdient thanks a lot

kirandesai303, 2018-10-23:
Put the "provided" instead of "CompileOnly"

schumannd, 2018-11-07:
After applying the changes I get:

```
Failed to apply plugin [id 'com.android.application']
   > Minimum supported Gradle version is 4.4. Current version is 4.1.
```

can anyone comment on whether updating to react-native 0.57 would solve this issue? Currently using 0.51

sibelius, 2018-11-07:
it could

TuzikY, 2019-04-29:
@ldstein

TuzikY, 2019-04-29:
@ldstein i use then iha
[image]
what should i do