react-native-camera/react-native-camera #1493: Crash on closing App

SimonErm, 2018-04-18:
### Which implementation are you using

RNCamera

### Actual behaviour
Crashes on clonsing App with Excception:
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'void org.reactnative.facedetector.RNFaceDetector.release()' on a null object reference
                                                                               at org.reactnative.camera.RNCameraView.onHostDestroy(RNCameraView.java:458)
                                                                               at com.facebook.react.bridge.ReactContext.onHostDestroy(ReactContext.java:230)
                                                                               at com.facebook.react.ReactInstanceManager.moveToBeforeCreateLifecycleState(ReactInstanceManager.java:690)
                                                                               at com.facebook.react.ReactInstanceManager.onHostDestroy(ReactInstanceManager.java:599)
                                                                               at com.facebook.react.ReactInstanceManager.onHostDestroy(ReactInstanceManager.java:613)
                                                                               at com.facebook.react.ReactActivityDelegate.onDestroy(ReactActivityDelegate.java:119)
                                                                               at com.facebook.react.ReactActivity.onDestroy(ReactActivity.java:72)
                                                                               at android.app.Activity.performDestroy(Activity.java:6425)
                                                                               at android.app.Instrumentation.callActivityOnDestroy(Instrumentation.java:1153)
                                                                               at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:4126)

### Environment
- **Node.js version**:
- **React Native version**:
- **platform version**: iOS 9.0, Android 5.1.1

### react-native-camera
**Version**: 1.1.0

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

SimonErm, 2018-04-18:
Wrong Repo. Belongs to cameraview

jgfidelis, 2018-04-18:
I think this is fixed on 1.1.1

closed 2018-04-18 by jgfidelis