react-native-camera/react-native-camera #1494: In my project,it crashes. with messge "RCTFatalException: Unhandled JS Exception: undefined is not an object (evaluating 'd.FaceDetection.Landmarks.none')".i added faceDetector by guide

06065044, 2018-04-18:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?

### Steps to reproduce
1.
2.
3.

### Does it work with Expo Camera?
RNCamera is based on Expo Camera, if this does not work with Expo Camera it won't work with RNCamera as well.
You should open an issue there as well, so we can cooperate in a solution.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**:
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

sibelius, 2018-04-18:
you linked wrong

try to follow docs

06065044, 2018-04-18:
thank you for your answer.if i install by cocoapods .should i call "react-native linked react-native-camera"?

closed 2019-04-18 by sibelius