react-native-camera/react-native-camera #1497: Android build failed

evscan, 2018-04-19:
### Which implementation are you using

RNCamera

### Steps to reproduce
1.
2.
3.

### Expected behaviour
Successful build

### Actual behaviour
Build failed
I attached error message below:
FAILURE: Build failed with an exception.

* Where:
Build file '...../node_modules/react-native-camera/android/build.gradle' line: 55

* What went wrong:
A problem occurred evaluating project ':react-native-camera'.
> Could not find method compileOnly() for arguments [com.facebook.react:react-native:+] on object of type org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

### Environment
- **Node.js version**: v8.9.4
- **React Native version**: 0.51.0
- **React Native platform + platform version**: Android

### react-native-camera
**Version**: master

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

sibelius, 2018-04-19:
upgrade to gradle 3

closed 2018-04-19 by sibelius

rena08, 2018-04-19:
I have installed the version 1.0.3 and my package.json looks like "react-native-camera": "^1.0.3". I delete the symbol ^ (before that I run: npm uninstall --save react-native-camera)  for it to install the version 1.0.3 and then It works fine.
It don't make you to upgrade graddle to version 3.