react-native-camera/react-native-camera #1500: Build fails with manual link IOS

connercms, 2018-04-19:
### Which implementation are you using
RNCamera

### Steps to reproduce
Follow manual link instructions exactly on readme
1. npm install react-native-camera --save
2. Right click libraries in Xcode project, add RNCamera.xcodeproj
3. Add libRNCamera.a to Build Phases - Link Binaries With Libraries (as a side note, libRNCamera.a is colored red, not sure how to resolve)
[image]
4. Ensured RNCamera project Header Search Paths contain both $(SRCROOT)/../../react-native/React and $(SRCROOT)/../../../React 
[image]
5. Clean and rebuild project 
[image]

### Expected behaviour
Project to build?

### Actual behaviour
Project does not build

### Environment
- **Node.js version**: 5.0.3
- **React Native version**: 0.51.0

### react-native-camera
**Version**: ^1.1.1

brenwell, 2018-04-20:
I have the identical issue, did you solve your issue?

connercms, 2018-04-20:
@brenwell No, I am still stuck on this problem

brenwell, 2018-04-20:
[image]

This fixed it for me Enable Modules -> **NO**

connercms, 2018-04-20:
I tried disabling this option for my project, RNCamera project, and both projects and build still failed every time.

connercms, 2018-04-20:
Right clicking on libRNCamera.a and clicking Show In Finder does not show anything, and libRNCamera.a text is red. I think it may have something to do with this but not very familiar with Xcode

brenwell, 2018-04-20:
Well that sucks, I also rolled back , perhaps that helps

```json
"react-native-camera": "1.0.3",
```

jgfidelis, 2018-04-20:
Is it working on 1.0.3?

jgfidelis, 2018-04-20:
Please try on 1.1.0 as well.

Maybe some dependency we introduced when adding CircleCI to the project caused this

connercms, 2018-04-20:
No. Have tried rolling back version, disabling modules as suggested above, removing library and reinstalling and trying react-native link, upgrading react-native version to latest and trying above combinations, but nothing works.

connercms, 2018-04-20:
Will try 1.1.0

connercms, 2018-04-20:
No luck. Disabled modules for RNCamera project removes the Parse Issue errors from the screenshot above, but can't get rid of the Lexical or Preprocessor Issue, 'string' file not found issue no matter what I try.

brenwell, 2018-04-23:
Problem returned for me and I couldn't shake it. Sadly I had to swap to an alternative lib. It appears the 1.0.3 didn't make any difference

closed 2018-04-24 by jgfidelis

connercms, 2018-04-24:
Closed with no solution? Looks like I'll find an alternative library also

FacuAcosta, 2018-04-25:
Having the same issue on v1.1.1, why is this even closed?

```
"react": "^16.3.2",
"react-native": "^0.55.3",
```

MrPistoul, 2018-05-04:
Same problem here
Someone have a solution ?
"react-native-camera": "^1.1.1"
"react": "^16.3.2",
"react-native": "^0.54.4",

himanshusoni, 2018-11-30:
Any updates on this issue ?

we recently moved from `react-native-barcode-universal` (which used rn-camera 1.0 internally ) to `react-native-camera: 1.4.3` 
The interesting fact is that we are facing the above issue on xcode 10. But its working fine on xcode 9.3.1 !!

```
    "react": "16.3.1",
    "react-native": "0.55.4",
```

FlavioNascimento10, 2019-07-10:
Same problem here. Someone have a solution?