react-native-camera/react-native-camera #1501: Face Detector doesn't work in "back" Camera

cristianBarreto, 2018-04-19:
### Implementation i'm using
*RNCamera*

### Steps to reproduce
1. Install de package: `npm i -S react-native-camera`
2. Link the package: `react-native link`
3. Follow this tutorial for solve problems with *gradle*: [Update Gradle](https://itnext.io/install-react-native-maps-with-gradle-3-on-android-44f91a70a395?gi=5162f7167b38)
4. Copy the *App.js*  from the [RNCamera Example](https://github.com/react-native-community/rncamera-example)
5. Run `react-native run-android`

### Does it work with Expo Camera?
Cloning the [Expo Camera example](https://github.com/expo/camerja/blob/master/App.js) it works fine.

### Expected behaviour
The face detector in 'back' camera should be work like in Expo Example

### Actual behaviour
The face detector in 'back' camera doesn't work like in Expo Example

### Environment
- **Node.js version**: 9.11.1
- **React Native version**: 0.55.3
- **React Native platform + platform version**:  Android 7.0

### react-native-camera
**Version**: 1.1.1

SyedKabeer, 2018-06-04:
Any update on this. I am stuck here.

yli18, 2018-07-26:
same here, I guess the problem is with orientation? Back camera only works after rotate 180 degree.
Also textRecognition has similar problem, 90, 180, 270 works fine, 0 degree doesn't work.

rzapalupi, 2018-08-06:
same problem with me, it doesnt work if portrait mode

ManigandanRaamanathan, 2019-02-08:
For me the front camera not detecting face. Any solution?

closed 2019-06-08 by stale[bot]