react-native-camera/react-native-camera #1503: Limit barcode scan area?

remisaado, 2018-04-20:
I am using the barcode scanner from react-native-camera and currently if I use it and there are multiple QR-codes closely on top of each other, I point my camera at one and it reads the code above it which is outside of the display on screen but within the cameras view. If however there is no QR-code above the one I want to scan, then it scans the correct one, so it seems like it always scans the top QR-code within the cameras view.

**Here's my question:** Is there a way to limit the "scan area" to be the same size and area as my camera view on my display?

```
<View style={styles.container}>
  <Camera
    style={styles.preview}
    onBarCodeRead={this.onBarCodeRead}
    ref={cam => this.camera = cam}
    aspect={Camera.constants.Aspect.fill}>
  </Camera>
  <Button
    style={styles.buttonStyle}
    <Text>{this.state.qrcode}</Text>
  </Button>
</View>

const styles = {
  container: {
    height: 300,
    flex: 1
  },
  preview: {
    flex: 1
  },
  buttonStyle: {
    marginTop: 20,
    marginLeft: 20,
    marginRight: 20,
    marginBottom: 20,
    alignSelf: 'center'
  }
}
```

Versions, if needed:
```
"react-native": "0.42.3",
"react-native-camera": "0.6.0",
```

[My question on Stackoverflow](https://stackoverflow.com/questions/49880395/react-native-camera-limit-barcode-scan-area)

YLeaves, 2018-04-28:
I also use this library to implement the QR code scanning function. The callback method of barcodeReceived is always called, so I set a recognition area based on the callback parameters. The callback parameter contains the frame you scanned. You can use a range of coordinate points as the recognition area, and then use a variable to mark the read information.

Here's my treatment. I hope I can help you

Ps: My QR code scan area is the size of 200x200 in the middle of the screen. The 2D code origin identification area is a range in the upper left corner

var width = Dimensions.get('window').width
var height = Dimensions.get('window').height

let viewMinX = (width - 200)/2;
let viewMinY = (height - 200)/2;

let isShowCode = false;

barcodeReceived(e) {
        //console.log(e)
        let x = Number(e.bounds.origin.x)
        let y = Number(e.bounds.origin.y)

        //识别二维码左上角坐标点位置
        if ((x > viewMinX && y > viewMinY )&&( x < (viewMinX + 60) && (y < viewMinY + 60))){
            if (!isShowCode){
                isShowCode = true;
                alert('Type: ' + e.type + '\nData: ' + e.data);
            }
        }
    }

twxia, 2018-06-11:
For iOS, I do something like @YLeaves says.

In Android, now always return finder pattern's coordinates. Also, I try to modify `RCTCameraViewFinder.java` and use `WhiteRectangleDetector` from https://github.com/zxing/zxing to get the boundary. I can't get the actual boundary of the qr code, but only finder pattern's coordinates. So it's hard to restrict the scanner area. Not sure I'm using the right method from `zxing`.

If we need to achieve this restriction area, seems we need to export some parameters(left, top, width, height) from `PlanarYUVLuminanceSource` to let developers to setup the scanning area.

HarshitMadhav, 2019-01-20:
Facing the same issue I am unable to set the custom width and height to the scan area.

anniewey, 2019-02-26:
@remisaado Did you manage to find any solution?

anniewey, 2019-06-03:
@gaodeng you can apply the `rectOfInterest` PR for iOS. But for Android I couldn't get it work with this library, hence using other [library](https://github.com/react-native-component/react-native-smart-barcode).

ntk0104, 2019-07-10:
any news guys?

jsastriques, 2019-07-19:
help here plz, i face the same doubt

mjm918, 2019-09-25:
Any update on this?

ceessay, 2019-11-03:
Having a similar use case. I am using text detection with ML Kit. Detection is made on electric meters which can have multiple text blocs on them. Would be useful to have a mask specifying the area to consider when performing detection, every piece of text outside that area would be ignored.

kaduelinho, 2019-11-19:
I'm having the same issue

lededje, 2019-11-21:
Can you use the bounding coords in the callback and use those to discard anything that is outside of the bounds you want?

MakhouT, 2020-01-06:
I also need this...

lededje, 2020-01-11:
I too had to do this recently. I've managed to achieve something I'm happy with so I thought I'd share it here. I believe the correct solution is a fully native one so I'm inclined not to make a pr to add this into the plugin but I'll share my workaround:

First I specified the height and width of the viewport I wanted to check against:

```js
const viewfinderHeight = 200;
const viewfinderWidth = 300;
```

You will need an aabbCollision function. These are pretty simple:

```js
const aabb = (obj1, obj2) => obj1.x < obj2.x + obj2.width
    && obj1.x + obj1.width > obj2.x
    && obj1.y < obj2.y + obj2.height
    && obj1.y + obj1.height > obj2.y;
```

All this does is check the height, width, x position and y position of two objects and see if they collide. This can be tweaked to ensure that one is entirely within another or maybe a prebuilt rotation-aware n-body continuous collision detection function but this wasn't needed for my project.

Next I needed to know which area of the screen was the area that is deemed safe.

```js
  const { height: windowHeight, width: windowWidth } = Dimensions.get('window');
  const viewFinderBounds = {
    height: viewfinderHeight,
    width: viewfinderWidth,
    x: (windowWidth - viewfinderWidth) / 2,
    y: (windowHeight - viewfinderHeight) / 2,
  };
```

Using the `onGoogleVisionBarcodesDetected` to get the the full bounding box, I checked each matched barcode against the viewfinder bounds.

```js
onGoogleVisionBarcodesDetected={({ barcodes: detectedBarcodes = [] }) => {
  const collidingBarcodes = barcodes.filter((barcode) => aabb(viewFinderBounds, {
    height: barcode.bounds.size.height,
    width: barcode.bounds.size.width,
    x: barcode.bounds.origin.x,
    y: barcode.bounds.origin.y,
  });

  setBarcodes(collidingBarcodes)
}
```
The last thing I needed was something visual so the user can see what area is designated for scanning. I used the react-native-barcode-mask and passed in the viewport sizes like so:

```jsx
<BarcodeMask width={viewfinderWidth} height={viewfinderHeight} showAnimatedLine={false} transparency={0.8} />
```

Below is an example of what I ended up with. The red barcode is not colliding with the viewport and is ignored.

[image]

closed 2020-04-26 by stale[bot]

kaduelinho, 2020-05-03:
Please don't close this issue, should be great if has this feature!!!

amirulasrof, 2020-05-04:
> Please don't close this issue, should be great if has this feature!!!

Now you can use rectOfInterest for both Android and iOS to limit scan area

skyrippa, 2020-10-21:
> > Please don't close this issue, should be great if has this feature!!!
> 
> Now you can use rectOfInterest for both Android and iOS to limit scan area

can you help me on how to implement it on Android? I can't find out what are the cameraViewDimensions or how to implement the percentages in rectOfInterest

shupingchu, 2020-10-22:
Is possible you could provide example for how to use `rectOfInterest` for both iOS and Android by chance? it seems like not working somehow..

amirulasrof, 2020-10-27:
Sorry for the late reply. Here is my sample code https://gist.github.com/amirulasrof/c6b64165f63baecea2fa53061eb2b80f @shupingchu @skyrippa

manuel3265, 2020-10-29:
How do I use it with the phone horizontally? this way it works for me with the device vertically

manuel3265, 2020-10-29:
@amirulasrof  How do I use it with the phone horizontally? this way it works for me with the device vertically

amirulasrof, 2020-10-30:
> @amirulasrof  How do I use it with the phone horizontally(landscape)? this way it works for me with the device vertically(portrait)

https://stackoverflow.com/questions/62472029/react-native-detect-screen-rotation-change-using-portrait-mode

You can refer solution above then update the scanArea accordingly

manuel3265, 2020-10-30:
@amirulasrof I have tried detecting the orientation change, but it does not scan

saadmoneylion, 2022-07-05:
@amirulasrof have you tried this

> Sorry for the late reply. Here is my sample code https://gist.github.com/amirulasrof/c6b64165f63baecea2fa53061eb2b80f @shupingchu @skyrippa

@amirulasrof have you tried this solution on Android? Is it working and limiting the scan Area? Because I am trying and it's not limiting the scanArea, I can still scan outside of my centered scan box.

LongVT8, 2022-10-18:
@amirulasrof 
 https://gist.github.com/amirulasrof/c6b64165f63baecea2fa53061eb2b80f

 I tried this solution on Android, but it not working and limiting the scan Area. Please check code .