react-native-camera/react-native-camera #1504: App crash when using tack picture on Samsung Galaxy A7 device

Fox-3394, 2018-04-20:
Hi every one, I has issue. 

### Description: 
When I using device Samsung Galaxy A7 tack picture, then app crash. I call logcat. and this result:

  
[image]
And file full log: 
[log-cat.txt](https://github.com/react-native-community/react-native-camera/files/1931749/log-cat.txt)

### Expected behaviour
Get uri image

### Actual behaviour
App crash

### Environment
- **React Native platform + platform version**: Android 7.0 - Samsung Galaxy A7 - Version 8.1

### react-native-camera
**Version**: v1.10

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

krystianpach, 2018-04-20:
Same on Samsung Galaxy s5 neo

mykelaballe, 2018-05-03:
is this fixed? i'm also experiencing it in samsung 4.2.2

closed 2019-06-08 by stale[bot]