react-native-camera/react-native-camera #1505: Chinese phones can not use google so app will crash

Eggplant-Cui, 2018-04-21:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using

RNCamera

### Steps to reproduce
RNCameraView.<init>

### Does it work with Expo Camera?
RNCamera is based on Expo Camera, if this does not work with Expo Camera it won't work with RNCamera as well.
You should open an issue there as well, so we can cooperate in a solution.

### Expected behaviour
Maybe add a switch or detactor 

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**:
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgfidelis, 2018-04-22:
Can you explain the title better?

cookorg, 2018-07-15:
我也是这么认为,在中国的手机默认没有google-play-services，很遗憾啊

pbmlly, 2019-04-17:
有什么替代方案吗？我最近也遇到这个问题了。

sibelius, 2019-04-18:
RCTCamera is removed https://github.com/react-native-community/react-native-camera/issues/1890

Migrate to latest version of RNCamera

closed 2019-04-18 by sibelius

yylingsj, 2021-06-12:
@sibelius 从3.43.x版本开始一直尝试，到现在的最新版3.44.1了，TextRecogized特性在安卓手机上依然会闪退，React Native使用的最新版0.64，iOS可以正常使用这个特性，无解了，还有人吗？