react-native-camera/react-native-camera #1506: Could not find method compileOnly

jsina, 2018-04-21:
I've follow the doc instruction but I've get the error for run android.
I've also followed this issue [workaround](https://github.com/react-native-community/react-native-camera/issues/1490) but I have still the problem. if there is a better step-by-step solution for resolving this problem please explain.

guguji5, 2018-04-22:
I follow the doc and run
```
npm install react-native-camera -save
react-native link
```
it occurs in my project.

closed 2018-04-22 by jgfidelis

timomeara, 2018-04-23:
why was this closed?
this is still happening today

FAILURE: Build failed with an exception.

* Where:
Build file '/Users/timo/git/router4/node_modules/react-native-camera/android/build.gradle' line: 55

* What went wrong:
A problem occurred evaluating project ':react-native-camera'.
> Could not find method compileOnly() for arguments [com.facebook.react:react-native:+] on object of type org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

guguji5, 2018-04-24:
@timomeara I have fixed this issue last weekend, I follow the most stared answer in the[link](https://github.com/react-native-community/react-native-camera/issues/1490). it works after I figure out several different errors.

p.s. my machine is mac, and I just try the android end.

jason-zuo, 2018-04-26:
i had this problem too

GetSource1234, 2018-05-14:
I have the same issue!

Bijiabo, 2018-06-11:
Same issue.
Can not understand why this package so hard to use.
Feel so bad.

sibelius, 2018-06-11:
check this diff https://github.com/entria/react-native-view-overflow/issues/25#issuecomment-396294644