react-native-camera/react-native-camera #1507: takePictureAsync not work properly on ios and crashed on Android due to Out of Memory

Deno028, 2018-04-23:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?
=> RNCamera

### Steps to reproduce
1. run takePictureAsync({ width: 1920, fixOrientation: true,quality:0.5 ,base64: true}) => get raw image from iPhone camera.
2. run takePictureAsync(options) function on android => crashed 
3.

### Does it work with Expo Camera?
I dont use expo

### Expected behaviour
Tell us what should happen
Should return object with the input options 

### Actual behaviour
Tell us what happens instead
- ios: return raw images not compressed
- android: crashed 

### Environment
- **Node.js version**:
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc
Environment:
  OS: macOS High Sierra 10.13.4
  Node: 7.3.0
  Yarn: 1.3.2
  npm: 5.7.0
  Watchman: 4.9.0
  Xcode: Xcode 9.3 Build version 9E145
  Android Studio: 2.2 AI-145.3360264

Packages: (wanted => installed)
  react: 16.2.0 => 16.2.0
  react-native: 0.53.0 => 0.53.0

ios: iphone 6s - ios 11.3
android: Asus zenphone - android 4.4.2

### react-native-camera
**Version**: npm version or "master"
1.0.1

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgfidelis, 2018-04-23:
Probably a problem with old android phones with less RAM, you could try a smaller width for android?

jpudysz, 2018-04-25:
Try smaller width and quality also clean (refactor) your build.gradle file. With all that settings we were able to take pictures on every Android.

Deno028, 2018-04-26:
Problem solved after update to 1.1.0

closed 2018-04-26 by sibelius

YouCii, 2018-09-07:
I lowered the width and solved the problem.

mschmuckler, 2018-12-27:
@YouCii Did you manage to find the maximum width that works on all androids? Or quality? Or combination of width/quality?

YouCii, 2018-12-28:
@mschmuckler I think it's basically impossible to find a stable configuration that works for all android. Maybe we can determine the current device's memory size to dynamically initialize these configurations.

Samykills, 2019-01-09:
ok so i think i found a solution, you gotta mention exif here like this : 
this.camera.takePictureAsync({ quality: 0.2, exif: false, base64: false })

mschmuckler, 2019-01-09:
@Samykills { exif: false, base64: false } are the defaults according to [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md).

Is quality: 0.2 good enough? That sounds incredibly low, as if the picture would be fuzzy. Also, if you don't specify a width, it seems you'd still be taking large pics (1200x + or something) with low quality.

Samykills, 2019-01-10:
@mschmuckler  yeah true these are defaults, but if i do not specify explicitly then this method is failing, not sure on the cause though.

And m keeping the quality low for my usecase as i need the same image blurred so it does'nt matter, hence a low sized file, have to provide a specific width here. thanks for the tip :D

csae0, 2019-04-03:
Add this to your manifest in the application tag:
android:hardwareAccelerated="true"
android:largeHeap="true"

worked for me

felansu, 2019-04-23:
@csae0 i go to test your solution

felansu, 2019-05-28:
work for my :D