react-native-camera/react-native-camera #1508: fixOrientation:true does't work on android

NoMoneyNoCar, 2018-04-23:
here is my code:
` <RNCamera
                    ref={(cam) => {
                        this.camera = cam;
                    }}
                    autoFocus={true}
                    playSoundOnCapture={true}
                    style={styles.container}
                    aspect={Camera.constants.Aspect.fill}

            ></RNCamera>`
and this is the callback method:
`
const options = { quality: 0.8,base64:true,width:1024,forceUpOrientation: true, fixOrientation: true};
this.camera.takePictureAsync(options)

                .then((data) => {
                     console.log(data);
                    if(this.props.navigation.state.params.needRecognize == true){
                        this.recognize(data);
                    }else{
                        if (this.props.navigation.state.params.callBack) {
                            this.props.navigation.state.params.callBack(data);
                        }
                    }
                })
                .catch(err => console.error(err))
                .finally(() => this.props.navigation.goBack());`
anyone has the same issue?

jgfidelis, 2018-04-23:
Can you post the base64 string generated by the camera here?

Small thing: aspect prop does not exist on RNCamera

NoMoneyNoCar, 2018-04-24:
jgfidelis:
          thank you for your answer,and here is the base64 string:

`"/9j/4AAQSkZJRgABAQAAAQABAAD/2wBDAAYEBQYFBAYGBQYHBwYIChAKCgkJChQODwwQFxQYGBcU↵FhYaHSUfGhsjHBYWICwgIyYnKSopGR8tMC0oMCUoKSj/2wBDAQcHBwoIChMKChMoGhYaKCgoKCgo↵KCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCj/wAARCAVVBAADASIA↵AhEBAxEB/8QAHwAAAQUBAQEBAQEAAAAAAAAAAAECAwQFBgcICQoL/8QAtRAAAgEDAwIEAwUFBAQA↵AAF9AQIDAAQRBRIhMUEGE1FhByJxFDKBkaEII0KxwRVS0fAkM2JyggkKFhcYGRolJicoKSo0NTY3↵ODk6Q0RFRkdISUpTVFVWV1hZWmNkZWZnaGlqc3R1dnd4eXqDhIWGh4iJipKTlJWWl5iZmqKjpKWm↵p6ipqrKztLW2t7i5usLDxMXGx8jJytLT1NXW19jZ2uHi4+Tl5ufo6erx8vP09fb3+Pn6/8QAHwEA↵AwEBAQEBAQEBAQAAAAAAAAECAwQFBgcICQoL/8QAtREAAgECBAQDBAcFBAQAAQJ3AAECAxEEBSEx↵BhJBUQdhcRMiMoEIFEKRobHBCSMzUvAVYnLRChYkNOEl8RcYGRomJygpKjU2Nzg5OkNERUZHSElK↵U1RVVldYWVpjZGVmZ2hpanN0dXZ3eHl6goOEhYaHiImKkpOUlZaXmJmaoqOkpaanqKmqsrO0tba3↵uLm6wsPExcbHyMnK0tPU1dbX2Nna4uPk5ebn6Onq8vP09fb3+Pn6/9oADAMBAAIRAxEAPwDGVR39↵OlPCgfWkUncc96coI75FYGSF4Pb6U+3wLjHYUg6f1p1tzcLgZ5/OhILmnEuBmhhjJHSnjABpgBPb↵igsdGOfep0XavvUcS/N0qzjoOKQ0RntRjinkDpSDgGgZWuhlSD39axJhjIGAMVtXp+XrWPMck4/O↵gCELgAdj71KpyuOwpg9GNKeKhMVhcdxTJMhueRT+n40Ljdnn370xj1xt/wAaCDyOcdaUep6Ui88f↵ypCY3GRUo4PNNA9SM09QRk9e9PcEPC561IuM4NCjvSqOfakMeOTkUpXNKOnFOA/+tRYEyLaM5FKV↵ycdQaeAOenpSgY7VVgPMPH3h77M5vIFPkyHLeimpPhz4lMEo0q+c4YgQux+vy16NdWqXdtJDMu6N↵xg14z4r0OXR79gNwUHdHIM1qtVYR7S5yCQckDNeA+MSW8RXpOcmQ16b4F8TDVbT7Jcti8iwDn+Ie↵teYeKfm8RXe48+d6+9KEbMLmbaHZdxHPRga9Xj5iXPO5e/fivLtRh8jUSi/7J/SvVrMebZQSDnKD↵pVS2BHn3ifSmsrougPlMcqf6VU0y7IkUnh0Nen3umxahaPBKoII4Poa8w1fTZ9LvWjkVgQcqccEe↵tC1QLQ9D0K/8yPa3IxyCKg1jwfDqG+exYQznkqfunj9K5TQ9VCOpc4Poa9E0a+V0HJPQ+9Qm4Mrc↵8yvtFv8AT5CtzbOBnhgMg8+tVEQu238697ivbRIGe52GMD5iwB4/GvMdUkg1PWJriCJIrZOFA4zW↵kJ8zIfulKwgWztWnlH7wjA+lY+17zVo4z8xZhkCu20/S01CCSW63LEBiNR396p2vhxbTU2uVdXTa↵SqnqDT51cuFNz23OYvQZ766Y8CM46+nFU0BkfA5FNkllMsqt1ZyW71p6TagnewORTirmV7GvoNhJ↵JNFbwKWmnIHTkDua9n03SI9N0JrWJQMRncfVsdaw/hz4fNvC1/coPNkGEBHKj1runjzC4wOlEn0K↵R8yrhNSXHab+Rr6EiYtZwt6ov8q+fNQzDqtwDxiVuv1r6BtOdMtjx/ql/kKzqDQRMAx6Vo2zYasU↵NhzWrbHOKzQzorE5xmtaIcCsfTugFbUA+U1Q7j+9OHU0hxmlHU4oEOXpS9qQdKXtRYLgelA6UHpQ↵DxSAQ0dqSkPAoGO/Gmk8U3NNY0AKaic0paonbg4oAp6rdJaWc9xIQEiQsc9OK+UdVu2v9QuLuQ/N↵PIX6epr3n4v6qLHwpNCrYkumEQGeo6n+VeGaXZtf6paWirkyyBSDzkVpTVlcUme3fC7Tf7O8KW5I↵IknJlb8a6xmy3FV7aNba2iij+6ihR26VKhyc1myi1F04qVWABqBXAWmNLweaTYkWPMAGaheUVXeb↵jrUBlyagZZkkHXpVZ5CSRmoixPek55oKQpY9KY1KeabnimIVhTenSlz1pOlAIM000E5FNJ560rAA↵PB9elNzSZxnJppbGTnihC2FJGTTSajaTGcdaYWJHNOwx7vye9Rsx+gowBmkHI5ppCuJilIGRmgcG↵jHWqExMc0tAHvTsfSgSGlfelxxR0696MZosMTAoHQilwBQODmgBAMexpBgZxSnnkUgFAhKXGKUcg↵0dQaEAgHc9aOvvinDpQKYIbjJzS9sUvtSdqQJgB2oAA6Uo6mkApgB5PtSGnAcehpCKEFxByeKAOc↵04Dg0nr1pABxigClxxiihIBOcnPSk4HU07GaQCgBMd6XqKUCkOeaYABn600CnDp70uMHrSEIPu4o↵7mlFGeeKAGjAzilHpRSgCgBuKAPTtSn9KAfwFMVwxnrSCl96McigBAMelO45o9c80YznFCdxiKMH↵FGOtL3NAyRz0o6ghuDzQQTjA5pRkelLnrihhY860/wAX6ZcEK8skLNx8y8fmK3rO6huEDQSxycAg↵qQf6153eeB75HxAySIeQQawpbHU9NbdJHPAVHUZX8yKrlTIse1qQe9OiAWcHHf1715Ppni/VLJgs↵rCdQeko5HtXVaT43sbiRVu0a3Pr1FLlsFj0MDApRxmq1jf2t6ita3EUuf7jAmrKkEnGMDrUlIlhX↵Az+lWOAOO1RQipscU9BojPBJNNcY7U/qCD601hxzUjKV1gKfWseY5ckGte9bapzXK6pqttZA+fKA↵c9M5NSnqIumQZ6/jTWlyc1yFx4vhQ/JExP16/hTIfF0DNh0K5p2GdksgNSxEE4Arn7DWra6GYnB5↵6E8itS3nDEGhXQjSXk4NGPm7Z60xHyM5pwzzkUkJD+9PUZwRUUfOeTiplwo5x9aBkyDAweaco6Gm↵pyM0/GOKSHccOp9qeo/IU1Bjjdn+lPA7ZqxCAfXPWlwM5/CjPPGPxpaABSAprL1/SotYsHt5QA2P↵kb+6a08Dr60d8imtAPBL+1vNA1XcjPHPC2VYDG4Vl300lzqH2iU/vJG3NXtfi3w/FrFozKAt0v3G↵x19q8W1WJ7W9aNx88ZwR71qndCNDxnb+TqkR7PCjZ/D/APVXb+FrvzdItweSAAT71yvie5ttW0zT↵7u1Y74l8qRTjKn/I/Wp/BeoBYntXYblO5PpzmpeqGel26oy9Kg1vQLXWLPypQBMB8knoaZp04dVA↵OPrWzA/TOMUouwWueHa5od3ot60Vwvf5XA4arGi67LaOElBK9jivZ9YtLC6snXUxEIBzvbHy/SvG↵dStrWLUZUsGZoA2AWq01MNjZ1PWJL9Vhi+VP4iDRpdt9olERO2BeXY9DWQreVGSq8fSqDTyjfhmA↵PUdqahZWQr33O7v9ftLCExRESMvAUVxuo6vc3smSxUZyAOKq29pNckCNCf5VdubFLOJUYhrhuqjn↵ApKCW4c3Yo2kBeTHf1r0TwF4d/te8jaRf9FhO5j6muc8N6NNqV4ltbKWkfkn0HrX0J4d0aPSbCK3↵hAGAMnHU1s/dRCVy5bW6xRhIwAqjAHpUjplCD0xVtIsLzXI/EzX5fD2hiS2IE8rbEPcepFc7Zoke↵A+JR5XiG9UZJEzD9a920iQNoFifWFfr0r57vJpLq7knkYvJIxYk9STXpPw/8VXFz5el3qgqiYice↵3aia0BHanmXB6dq1rT+GskH5s981pWbDcMc1mhnT6cRuHc4rah6GsHTWB71twt8tUOxL3p696jzz↵TlPFAiSgnimg5Whj8tACk8UhNNPSkLZpAOBprGmk00txQAu6mseKaTTC3XNACluKglfCnnFKzYFZ↵2p3aWlnPcTMBHEpc5OOlAHjXxi1P7d4iis423Jap8wB43H/AVV+E2nfa/EE18+THbJtXj+I8fyrl↵9Uvpb27vb+YkvM5b9TXq3w3sTpnhqJpRiW4/et+PStG+WIt2dqzgD1NCSADms83GW4pyyEisEyy+↵8wCnkYqAvnvxUBY4xS/XtQBKx4603pSE4HWhT60ALkdaQnrikPSm9qQIXJAOKQ9KKaWzQhsAcD60↵meDSE4J5zUbOB1NNCHZ55NNLAVE8v93OajYkimkFyR5B0HNM3bup6Ug4HNNGaaQhwopB1xS/zp2E↵B5zSgYWkFLjjrxQAgowe2KOOxoAoAXgcmgc0nsaB6igAPPFKKMZHXFA6U7gBpDTjjFJ1qUAn0oYY↵FLS5yKbGNHSgcZpQMUcUCEFLwBSDrmnfyFMBq+vel+lHGeKX1GaQhoGKUdaUAUDkmhjEPNJjnmlA↵60YxQADrik2nNOxS9jTERkc0EGnEE0gGM+5oQIU8UmOM0pOcfyoHT1oYxO9DfSjjmigQvFJx0oAx↵S4yOtUAh6cUAcZ/Ol6jFL60mA1RxR2xinDg0h47k0IGNxxRjFOHSjtRYVhozj296UE8Uvr2o7Ckk↵FgxSD3p2eKb3oQxB3zSjgH1ox19aFz0PWmJAcUnRaXsaOxGeaGMwQoA459KGiV1wwDL3B5pBLGv8↵Yx9aQXMIz+8X86V2QZWoeFdLvVOYfKbpmPj6VympeA5oiTZTiXP8JG2vQReQY5mQfVgKb9rtlmU+↵fFlT13A8VUZPqJHkNzp2raLMS0c9u399ehH+cVuaL8QdSs8R3qLcxgjkjawH1FesTS2U8WyaW2kU↵9mYEVy+reD9H1IbrZ0t5f+mZBH5UXRaNfw/4v0nVVURXAimP/LKX5T+HrXTA5HHcV4LrXgvV…i38NWUgMF1Gx79a0oo1VBtAFV7/AkiJzj↵PNMENvW3KabZtjp1qO5PGBSWZw554pxEzeZh9l6jgVmbvzrRQk2Z5IOMVlggOfbrTBEoxzSgk5H9↵aYpFPDYzgZpjJrMkXERH94V714QbOmQ/SvA4uJUPOcivd/Bbk6XH7CplsNHVR+tWIzmqsZqzFWTK↵LSGpV5qCNuMVMlAiQHFOBzTB0pwoAguhmI5qTTn3wYz04plwPlINRaW+JJEP1FJDLswPJqBDtNWp↵BkGqbjmhCNOBsgVNVO0birnQVQx8Z4NSqagQ8VKlMRZhOGqxtzxVEHmrMMvIB/OmmBaWpE600IcA↵9qco5oAM81Vuz84q2+yJS0jY9qz7mYSyZUYUdKQEsHSpxVaA8GrCnpQAMoYYNVZYSvTkVbpKAM/F [... cut ...] 73HqaKKa3Ei5p4x6evTpUdz/rTj1ooqkMQcg0IdyA45oooiDFU5JHTtXsHw0JOm7fSiipYkd5D3↵q1H2oorEonjPWrKGiimxj1PBqZaKKBDqJhlCD6UUUmM5y/8A3chdeqnNdPZuXgRj1IoooWwCXK8G↵qsBwaKKECNOM5UVKDzRRVIT2JV6UooooQ0SRnDrikveRRRR0EQk8Coz0oooAdAfnNX4z8tFFOID1↵6mlFFFAxw6U7tRRUiEzxThRRVMBRTs0UU0Ao6Ux1DDmiigEVmGM0yiihAKDijNFFMBR3oz0oooAc↵KAaKKBj8Ug6UUUCQope9FFADhR6UUUALS5oooAWgUUUAKDzS0UUAKKO9FFACilHeiigApaKKACjt↵RRQN7DgOKO1FFAgoPFFFCBAOlFFFIAHFHaiimAvakoooGBpe1FFAgFA60UUIa3EzTh0oooEJRRRQ↵NCiloooEJQOlFFAC96bRRQAo70x6KKAP/9k=↵"`

jgfidelis, 2018-04-24:
This does not help me :/ `↵` I think this characters are messing up the online viewer that I am using. I just wanted to take a look into the photo using this website:
https://codebeautify.org/base64-to-image-converter

pstanton, 2018-05-03:
related: the exif PixelXDimension PixelYDimension are switched if fixOrientation:true

BigPun86, 2018-09-20:
fixOrientation:true doesn't work for me :(

Shabnam-Veranloo, 2019-01-14:
fixOrientation:true doesn't work for me when guality is 1 (if quality==0.5 its work) :(

closed 2019-06-08 by stale[bot]