react-native-camera/react-native-camera #1509: Stop preview in RNCamera

Shauneoo, 2018-04-23:
### Which implementation are you using
RNCamera

### Expected behaviour
this.camera.stopPreview() to stop preview screen

### Actual behaviour
this.camera.stopPreview() not a function in the latest RNCamera. Is there a work around of this? I have several instances of RNCamera in my app and trying to reduce app impact on performance.

### Environment
- **Node.js version**: v8.5.0
- **React Native version**:  "0.52.0",
- **React Native platform + platform version**: iOS 11.0

### react-native-camera
**Version**: "master"

jgfidelis, 2018-04-23:
You can test this pr: #1442

Shauneoo, 2018-04-23:
@jgfidelis thanks for the, will do. Forgive me but how do I get the specific link to that pull request? 'https://github.com/react-native-community/react-native-camera/pull/1442' doesn't seem to work.

mstange22, 2018-05-18:
Not sure if this is the same issue, but I'm looking at implementing a camera (front-facing) in our app to record facial emotions, but I don't want to display the preview.  Is there a way to disable the display to save real estate and resources (working in Android)?

YouCii, 2018-08-07:
My android app runs very slowly while navigating back from a camera screen.
So i need to stop preview before navigation.

AnurajDev, 2019-02-18:
Are there any specific reasons why "stopPreview()" is not implemented in RNCamera? I have the same problem @YouCii  has.

n1ru4l, 2019-02-18:
Despite the fact that you can simple unmount the camera component when you navigate away from a screen (e.g. react-navigation https://reactnavigation.org/docs/en/with-navigation-focus.html), there is a `pausePreview` method: https://github.com/react-native-community/react-native-camera/blob/808fbb1be06b4f19f7653246ebb6ffdc35e9315e/src/RNCamera.js#L454

@AnurajDev Is there anything not working with that method?

n1ru4l, 2019-02-18:
Do a conditional render and return null instead of the RNCamera in case the property injected by WithNavigationFocus is false

AnurajDev, 2019-02-19:
@n1ru4l  Thanks a lot for the replies. 

My problem was my device time and computer time were out of sync. After they were synced, navigation is happening much faster. 

Thanks again. :)

closed 2019-02-19 by n1ru4l