react-native-camera/react-native-camera #1512: Use Google Mobile Vision to scan barcode to improve.

ducpt2, 2018-04-24:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?

### Steps to reproduce
 Use RNCamera scan barcode (blur or not enough sharpen image barcode) slower than use Google Barcode Vision (i test on both android and ios).

### Expected behaviour
 Faster detect barcode.

### Actual behaviour
Slower (or CAN'T detect withblur or not enough sharpen image barcode )

### Environment
- **React Native version**: 0.53.3
- **React Native platform + platform version**: iOS 10.0, Android 6.0, etc

### react-native-camera
git+https://git@github.com/react-native-community/react-native-camera

jgfidelis, 2018-04-24:
So, are you saying that using GMV somewhere else it detects barcodes faster than the GMV in the RNCamera?

ducpt2, 2018-04-26:
@jgfidelis you mean RNCamera current using GMV to detects barcodes? 
i'm using => https://github.com/ekreutz/react-native-barcode-scanner-google for android 
 and => https://github.com/googlesamples/ios-vision for test ios.
Both of them detect faster (or RNCamera can not detect blur barcode) than using RNCamera example => https://github.com/react-native-community/rncamera-example.
@jgfidelis Maybe i miss the new version of RNCamera? Please help. Thanks.

jgfidelis, 2018-04-26:
I get what you mean now.

How much faster, do you have any time comparison just from looking?

ducpt2, 2018-04-27:
@jgfidelis if you notice, i have said CAN NOT detect blur barcode (barcode image has a poor quality). If you want, i can send you my image of barcode for testing?

danielsantiago, 2018-05-16:
I also had notice that GMV is much faster detecting barcodes also I'm able to scan all these bar codes with GMV: https://github.com/bees4honey/mobile_vin_scanner/tree/master/VIN%20Examples. But have problem scanning these two with RNCamera:

https://github.com/bees4honey/mobile_vin_scanner/blob/master/VIN%20Examples/7.JPG
https://github.com/bees4honey/mobile_vin_scanner/blob/master/VIN%20Examples/5.JPG

sibelius, 2018-05-16:
We have GMV for barcode scanning as well on RNCamera

check the code here https://github.com/react-native-community/react-native-camera/blob/master/src/RNCamera.js#L82

reopen if it is not working well

danielsantiago, 2018-05-16:
@sibelius but RNCamera already support GMV barcode in iOS too?

sibelius, 2018-05-16:
we gonna accept a pr to implement this on iOS side

sibelius, 2019-04-18:
@dov11 made a PR to use MLKit for this https://github.com/react-native-community/react-native-camera/pull/2209

closed 2019-04-18 by sibelius