react-native-camera/react-native-camera #1513: Photo orientation is wrong on Samsung S4

ziks21, 2018-04-24:
### Which implementation are you using
*RNCamera*

### Steps to reproduce
1. Make photo on Samsung device

### Expected behaviour
Photo image must be in correct orientation

### Actual behaviour
Photo image is rotated 90 degrees

### Environment
- **Node.js version**: 6.4.0
- **React Native version**: 0.48.4
- **React Native platform + platform version**: Android 5.0.1 (Samsung S4)

### react-native-camera
**Version**: 1.1.1

Hello. Thank you for this awesome package. 
Photo orientation works incorrectly on Samsung S4 device. Param fixOrientation true usage doesn't solve the problem. This issue might be related to generic problem with Samsung devices described here:
https://stackoverflow.com/questions/27515487/android-camera-picture-orientation-issues-with-samsung-galaxy-s3-s4-s5
https://stackoverflow.com/questions/11023696/setrotation90-to-take-picture-in-portrait-mode-does-not-work-on-samsung-device

But if i try to rotate image according to Exif Orientation property - it breaks orientation on some other devices, e.g. Sony Experia Z3 Compact Android 6.0.1. 

Do you have any idea how to fix that in some generic way? 
Thank you

PS: There is similar bug ticket that is closed by some reason, i wrote there - but no any feedback, that's why i decided to open new one, sorry for dupplicating: https://github.com/react-native-community/react-native-camera/issues/1348

jgfidelis, 2018-04-24:
Thanks for the research. Android is weird

bvanleeuwen1995, 2018-04-26:
This appears to affect all Samsung devices that we have tested, also the exif data is only passed to the promise callback, is it possible to add it to the image file also?

ziks21, 2018-04-26:
It might be that i confused forceUpOrientation param with fixOrientation, sorry. I will recheck if fixOrientation param fixes problem on Samsung S4 this week

jgfidelis, 2018-04-26:
@ziks21 nice observation, let us know.

ziks21, 2018-05-08:
@jgfidelis I can confirm that using fixOrientation true param photo has correct orientation on Samsung S4

mun3, 2018-06-05:
Do you guys have the same problem on the Android Emulator though?

chicotech, 2018-08-01:
I got the same problem on Nexus 5x and Pixel XL(Emulator). I tried fixOrientation=true and nothing change.

Sterlingryan, 2018-08-17:
I still have this bug, even after I add fixOrientation to true

fozzarelo, 2018-09-05:
is still an issue? it would mean the library is not compatible with several devices, if that's the case there should be a big warning on top of the readme.

doosti, 2018-09-09:
@ziks21 thanks. this solve my problem.
just set in option   fixOrientation:true 
my phone is LG k10

32penkin, 2018-11-01:
Samsung J7 - the same issue

KryDos, 2019-03-15:
@chicotech in case of Nexus and Pixel emulator I've noticed that `ExifInterace` cannot parse Orientation tag from the image data. Other tags parsed fine though. I don't think it is the issue of this library because library expects correct Orientation tag. If I set it manually then image is rotated properly. 

Were you able to solve this issue on your end?