react-native-camera/react-native-camera #1515: RNCamera takePictureAsync not trigger when reload

top24056, 2018-04-24:
### Which implementation are you using

*RNCamera*

Here's my code
```
takePicture = async function(){
        if(this.camera){
            const options = {
                quality : 1,
                width : 1000,
                height : 1000,
                fixOrientation : true,
                base64 : true
            };
            console.log('ready')
            const data = await this.camera.takePictureAsync(options);
            this.props.ImageAction(data);

            
            this.props.navigation.navigate("Modal");
            
            // console.log("store image = ",this.props.image.image_food)
        }
    }
```
### Steps to reproduce
1. When I edited code react native hot reload will reload app
2. takePicture will not trigger 
3. I have to rebuilt my app 

What happened?? I don't want to rebuilt my app every time I edited my code in takePicture() 
What can I do??

### Does it work with Expo Camera?
RNCamera is based on Expo Camera, if this does not work with Expo Camera it won't work with RNCamera as well.
You should open an issue there as well, so we can cooperate in a solution.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**: v8.9.4
- **React Native version**: react-native-cli: 2.0.1
- **React Native platform + platform version**: Android 8.0, etc

jgfidelis, 2018-04-24:
I think this is more of React-Native bug than RNCamera's...

jgfidelis, 2018-04-24:
Is the method not being called or `this.camera` is being set to null and when the `takePicture` is called it does not enter the if?

leandro-a-silva, 2018-04-25:
I also have the same problem here.

ayushmankd, 2018-05-25:
Same issue with me!

amitkumarsingh1991, 2018-05-30:
Hey @ayushmankd , what permissions u added in manifest file , please get back to me

ayushmankd, 2018-06-27:
Camera Permissions, Read and Write Permissions

closed 2019-06-08 by stale[bot]