react-native-camera/react-native-camera #1516: RCTFatalException: Unhandled JS Exception: undefined is not an object (evaluating 'f.FaceDetection.Landmarks.none')

karltaylor, 2018-04-24:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?

I am using `RNCamera`.

### Steps to reproduce
1. `npm install --save react-native-camera`

2. Run script to remove FaceDetection (Since it's not working for me because of this bug)

```bash
#!/bin/bash
echo "Creating project without FaceDetector"
if [ -e node_modules/react-native-camera/ios/FaceDetector ] ; then
  rm -rf node_modules/react-native-camera/ios/FaceDetector
fi
cp node_modules/react-native-camera/postinstall_project/projectWithoutFaceDetection.pbxproj node_modules/react-native-camera/ios/RNCamera.xcodeproj/project.pbxproj
```

3. Remove `libRTCCamera.a` from "build phases" like it says to [here](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md#ios) 

### Does it work with Expo Camera?
N/A

### Expected behaviour

App should not terminate for something being undefined.

### Actual behaviour

App terminates on unhandled JS Exception.

### Environment
- **Node.js version**: `v8.10.0`
- **React Native version**: `v0.50.4`
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc: `iOS 9.0`

### react-native-camera
**Version**: npm version or "master": `v1.1.1`

karltaylor, 2018-04-25:
Closing because I fixed it and I don't know how.

Tried again and did the following:

1. Remove `react-native-camera` from package.json.
2. Run `npm install --save react-native-camera // this will install latest from npm`.
3. Run remove face detection script.
4. Remove `LibRTCCamera.a` from Target -> Build Phases -> Link Binaries With Libraries
5. Product -> Clean (Shift + Cmd + K)
6. Run on device.

closed 2018-04-25 by karltaylor

webexpert4rv, 2019-04-23:
Not working, could you please clarify it more?