react-native-camera/react-native-camera #1527: Low resolution photos on Google Pixel with RNCamera

almost, 2018-04-28:
Sorry this is a vague one but I have a customer using my app on the Google Pixel and after the upgrade from RCTCamera to RNCamera the resolution of the photos has dropped from 4048x3036 to 720x480. I'm not specifying a width but I thought that was meant to mean I got the highest resolution possible, instead I seem to be getting (at least close to) the lowest. The weirdest thing is that it works fine on Samsung phone and tablet and my older Android phone.

Here's the bit of the code that uses the camera view:

```
      <RNCamera
        style={this.props.style}
        ratio={ratio}
        type={this.props.frontCamera ? "front" : "back"}
        flashMode={this.props.flashMode}
        onLayout={this._onLayout}
        onCameraReady={this._cameraReady}
        ref={this._gotRef}
      />
```

Ratio in this case is "4:3".

And the code to do the capture:

```
this.camera.takePictureAsync({mirrorImage: this.props.frontCamera});
```

Any ideas where to look to track this down?

almost, 2018-04-30:
Thanks João! The problem I'm having is that it doesn't happen on my test devices just my customer's device. So I'm really limited in what I can do to debug. I can get them to try a few things but I can't keep on bothering them.

I'm pretty sure the file I'm seeing is directly the one returned by react-native-camera. I store the original version before doing any processing.

jgfidelis, 2018-04-30:
Yup, the problem with Android a lot of times is exactly this, you can test and cover every edge case that you find in your test devices, but there will always be a different device with a strange bug and you cannot test or debug this.

almost, 2018-04-30:
Good times :)

Actually I've realized the ratio might not be set to 4:3. I'm guessing 4:3 is the shape of the sensor on most phones so to capture the most image that's probably the one to use. Know if that's right?

n1ru4l, 2018-11-06:
@matamicen Would be good to know the version number of the last working version!

matamicen, 2018-11-06:
@n1ru4l Sorry ... here is the information where everything was ok:

    "react": "16.4.1",
    "react-native": "0.56.0",
    "react-native-camera": "^1.1.4"

And this is the actual versions where the resolution go down:

    "react": "16.6.0-alpha.8af6728",
    "react-native": "0.57.4",
    "react-native-camera": "^1.3.1"

Hope this helps ... Thanks for your help.

closed 2019-06-08 by stale[bot]

GaspardC, 2020-05-06:
I have the same problem on a Galaxy S20 with the camera app I get 40MPx photos while in my app I manage to get only 12MPx