react-native-camera/react-native-camera #1528: Argument list too long: recursive header expansion failed at

M1chaelChen, 2018-04-29:
### Which implementation are you using
RNCamera

### Steps to reproduce
1. `npm i react-native-camera --save`
2. react-native link/manual link
3. build from xcode

### Expected behaviour
Successfully build the app

### Actual behaviour
Error when building:
Argument list too long: recursive header expansion failed at .../APP/node_modules/react-native-camera/ios/../../../ios/Pods/React/Libraries.

### Environment
- **Node.js version**: 8.10.0
- **React Native version**: 0.55.3
- **React Native platform + platform version**: IOS 11

### react-native-camera
**Version**: 1.1.2

Xaber20110202, 2018-04-30:
Same issue

Xaber20110202, 2018-05-15:
Hi, @M1chaelChen  I tried every methods, and still not work. Every time I removed `theProject/ios/DerivedData`, it built succeed. But when I closed xcode, still this error.

It puzzled me for a quite long time, after today I tried move `~/somedir/somedir/theProject` to `~/theProject`, this error never occur again.

shaikhussian, 2018-06-15:
@Xaber20110202 @M1chaelChen 
I am facing an error like this, i am unable to fix this, can any one give some suggestions about how to fix.

Argument list too long: recursive header expansion failed at /Users/nueve/Desktop/waitcheckbackup/node_modules/react-native-firebase-analytics/ios/../../../ios/Pods/React/node_modules/resolve/test/resolver/symlinked.