react-native-camera/react-native-camera #1532: com.android.support:support-v4 version conflict

rexlow, 2018-05-02:
### Actual behaviour
I am using this library together with `react-native-firebase`. Installation for iOS with pods went fine, it was a disaster on Android. I followed the official installation instruction for both automatic linking and manual linking and both failed with this error.

```
Android dependency 'com.android.support:support-v4' has different version for the compile (26.1.0) and runtime (27.1.0) classpath. You should manually set the same version via DependencyResolution
```

I tried to mitigate this error with the command below [(similar issue on Flutter),](https://github.com/flutter/flutter/issues/14020) but it just gets worse. 

```
subprojects {
    project.configurations.all {
        resolutionStrategy.eachDependency { details ->
            if (details.requested.group == 'com.android.support'
                && !details.requested.name.contains('multidex') ) {
                        details.useVersion "26.1.0"
            }
        }
    }
}
```

I've reviewed all related files and do not see any that specifies `26.1.0`. Here's my configs if they help.

**Manifest**

```
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.smarthome">

    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW"/>
    <uses-permission android:name="android.permission.RECEIVE_BOOT_COMPLETED" />
    <uses-permission android:name="android.permission.VIBRATE" />

    <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />

    <application
      android:name=".MainApplication"
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:allowBackup="false"
      android:theme="@style/AppTheme">
      <activity
        android:name=".MainActivity"
        android:launchMode="singleTop"
        android:label="@string/app_name"
        android:configChanges="keyboard|keyboardHidden|orientation|screenSize"
        android:windowSoftInputMode="adjustResize">
        <intent-filter>
            <action android:name="android.intent.action.MAIN" />
            <category android:name="android.intent.category.LAUNCHER" />
        </intent-filter>
      </activity>
      <activity android:name="com.facebook.react.devsupport.DevSettingsActivity" />

      <!-- RNFirebase messaging -->
      <service android:name="io.invertase.firebase.messaging.RNFirebaseMessagingService">
        <intent-filter>
          <action android:name="com.google.firebase.MESSAGING_EVENT" />
        </intent-filter>
      </service>
      <service android:name="io.invertase.firebase.messaging.RNFirebaseInstanceIdService">
        <intent-filter>
          <action android:name="com.google.firebase.INSTANCE_ID_EVENT"/>
        </intent-filter>
      </service>
      <service android:name="io.invertase.firebase.messaging.RNFirebaseBackgroundMessagingService" />
    </application>

</manifest>
```

**Project Level Gradle File**

```
// Top-level build file where you can add configuration options common to all sub-projects/modules.

buildscript {
    repositories {
        jcenter()
        google()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:3.1.2'
        classpath 'com.google.gms:google-services:3.1.2'
    }
}

allprojects {
    repositories {
        mavenLocal()
        jcenter()
        google()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }

    }
}
```

**App Level Gradle File**

```
apply plugin: "com.android.application"

import com.android.build.OutputFile

project.ext.react = [
    entryFile: "index.js"
]

apply from: "../../node_modules/react-native/react.gradle"

def enableSeparateBuildPerCPUArchitecture = false

def enableProguardInReleaseBuilds = false

android {
    compileSdkVersion 23
    buildToolsVersion "27.0.3"

    defaultConfig {
        applicationId "com.smarthome"
        minSdkVersion 16
        targetSdkVersion 22
        versionCode 1
        versionName "1.0"
        ndk {
            abiFilters "armeabi-v7a", "x86"
        }
    }
    splits {
        abi {
            reset()
            enable enableSeparateBuildPerCPUArchitecture
            universalApk false  // If true, also generate a universal APK
            include "armeabi-v7a", "x86"
        }
    }
    buildTypes {
        release {
            minifyEnabled enableProguardInReleaseBuilds
            proguardFiles getDefaultProguardFile("proguard-android.txt"), "proguard-rules.pro"
        }
    }
    // applicationVariants are e.g. debug, release
    applicationVariants.all { variant ->
        variant.outputs.each { output ->
            // For each separate APK per architecture, set a unique version code as described here:
            // http://tools.android.com/tech-docs/new-build-system/user-guide/apk-splits
            def versionCodes = ["armeabi-v7a":1, "x86":2]
            def abi = output.getFilter(OutputFile.ABI)
            if (abi != null) {  // null for the universal-debug, universal-release variants
                output.versionCodeOverride =
                        versionCodes.get(abi) * 1048576 + defaultConfig.versionCode
            }
        }
    }
}

dependencies {
    implementation project(':react-native-camera')
    implementation(project(':react-native-firebase')) {
        transitive = false
    }
    implementation "com.google.firebase:firebase-core:15.0.0"
    implementation "com.google.android.gms:play-services-base:15.0.0"
    
    implementation "com.google.firebase:firebase-auth:15.0.0"
    implementation "com.google.firebase:firebase-database:15.0.0"
    implementation "com.google.firebase:firebase-storage:15.0.0"
    implementation "com.google.firebase:firebase-messaging:15.0.0"
    implementation "me.leolin:ShortcutBadger:1.1.21@aar"
    
    implementation project(':react-native-fs')
    implementation project(':react-native-vector-icons')
    implementation fileTree(dir: "libs", include: ["*.jar"])
    implementation "com.android.support:appcompat-v7:23.0.1"
    implementation "com.facebook.react:react-native:+"  // From node_modules
}

// Run this once to be able to run the application with BUCK
// puts all compile dependencies into folder libs for BUCK to use
task copyDownloadableDepsToLibs(type: Copy) {
    from configurations.compile
    into 'libs'
}

apply plugin: 'com.google.gms.google-services'
```

**MainApplication.java**

```
package com.smarthome;

import android.app.Application;

import com.facebook.react.ReactApplication;
import org.reactnative.camera.RNCameraPackage;
import io.invertase.firebase.RNFirebasePackage;
import io.invertase.firebase.auth.RNFirebaseAuthPackage;
import io.invertase.firebase.storage.RNFirebaseStoragePackage;
import io.invertase.firebase.database.RNFirebaseDatabasePackage;
import io.invertase.firebase.messaging.RNFirebaseMessagingPackage;
import io.invertase.firebase.notifications.RNFirebaseNotificationsPackage;
import com.oblador.vectoricons.VectorIconsPackage;
import com.facebook.react.ReactNativeHost;
import com.facebook.react.ReactPackage;
import com.facebook.react.shell.MainReactPackage;
import com.facebook.soloader.SoLoader;

import java.util.Arrays;
import java.util.List;

import com.rnfs.RNFSPackage;

public class MainApplication extends Application implements ReactApplication {

  private final ReactNativeHost mReactNativeHost = new ReactNativeHost(this) {
    @Override
    public boolean getUseDeveloperSupport() {
      return BuildConfig.DEBUG;
    }

    @Override
    protected List<ReactPackage> getPackages() {
      return Arrays.<ReactPackage>asList(
          new MainReactPackage(),
            new RNCameraPackage(),
            new RNFirebasePackage(),
            new RNFirebaseAuthPackage(),
            new RNFirebaseStoragePackage(),
            new RNFirebaseDatabasePackage(),
            new RNFirebaseMessagingPackage(),
            new RNFirebaseNotificationsPackage(),
            new VectorIconsPackage(),
            new RNFSPackage()
      );
    }

    @Override
    protected String getJSMainModuleName() {
      return "index";
    }
  };

  @Override
  public ReactNativeHost getReactNativeHost() {
    return mReactNativeHost;
  }

  @Override
  public void onCreate() {
    super.onCreate();
    SoLoader.init(this, /* native exopackage */ false);
  }
}
```

### Environment
- **Node.js version**: 8.3.0
- **React Native version**: 0.54.2

### react-native-camera
**Version**:  1.1.2

jgfidelis, 2018-05-02:
change `implementation project(':react-native-camera')` to:
```
implementation (project(':react-native-camera')) {
        exclude group: "com.android.support"
    }
implementation "com.android.support:exifinterface:+"
  implementation "com.android.support:support-annotations:+"
  implementation "com.android.support:support-v4:+"
```

rexlow, 2018-05-02:
As soon as I rebuild, it gives another error (AAPT2) and a few other errors.

```
org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':app:processDebugResources'.
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeActions(ExecuteActionsTaskExecuter.java:100)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.execute(ExecuteActionsTaskExecuter.java:70)
	at org.gradle.api.internal.tasks.execution.OutputDirectoryCreatingTaskExecuter.execute(OutputDirectoryCreatingTaskExecuter.java:51)
	at org.gradle.api.internal.tasks.execution.SkipUpToDateTaskExecuter.execute(SkipUpToDateTaskExecuter.java:62)
	at org.gradle.api.internal.tasks.execution.ResolveTaskOutputCachingStateExecuter.execute(ResolveTaskOutputCachingStateExecuter.java:54)
	at org.gradle.api.internal.tasks.execution.ValidatingTaskExecuter.execute(ValidatingTaskExecuter.java:60)
	at org.gradle.api.internal.tasks.execution.SkipEmptySourceFilesTaskExecuter.execute(SkipEmptySourceFilesTaskExecuter.java:97)
	at org.gradle.api.internal.tasks.execution.CleanupStaleOutputsExecuter.execute(CleanupStaleOutputsExecuter.java:87)
	at org.gradle.api.internal.tasks.execution.ResolveTaskArtifactStateTaskExecuter.execute(ResolveTaskArtifactStateTaskExecuter.java:52)
	at org.gradle.api.internal.tasks.execution.SkipTaskWithNoActionsExecuter.execute(SkipTaskWithNoActionsExecuter.java:52)
	at org.gradle.api.internal.tasks.execution.SkipOnlyIfTaskExecuter.execute(SkipOnlyIfTaskExecuter.java:54)
	at org.gradle.api.internal.tasks.execution.ExecuteAtMostOnceTaskExecuter.execute(ExecuteAtMostOnceTaskExecuter.java:43)
	at org.gradle.api.internal.tasks.execution.CatchExceptionTaskExecuter.execute(CatchExceptionTaskExecuter.java:34)
	at org.gradle.execution.taskgraph.DefaultTaskGraphExecuter$EventFiringTaskWorker$1.run(DefaultTaskGraphExecuter.java:248)
	at org.gradle.internal.progress.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:336)
	at org.gradle.internal.progress.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:328)
	at org.gradle.internal.progress.DefaultBuildOperationExecutor.execute(DefaultBuildOperationExecutor.java:199)
	at org.gradle.internal.progress.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:110)
	at org.gradle.execution.taskgraph.DefaultTaskGraphExecuter$EventFiringTaskWorker.execute(DefaultTaskGraphExecuter.java:241)
	at org.gradle.execution.taskgraph.DefaultTaskGraphExecuter$EventFiringTaskWorker.execute(DefaultTaskGraphExecuter.java:230)
	at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker.processTask(DefaultTaskPlanExecutor.java:123)
	at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker.access$200(DefaultTaskPlanExecutor.java:79)
	at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker$1.execute(DefaultTaskPlanExecutor.java:104)
	at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker$1.execute(DefaultTaskPlanExecutor.java:98)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionPlan.execute(DefaultTaskExecutionPlan.java:626)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionPlan.executeWithTask(DefaultTaskExecutionPlan.java:581)
	at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker.run(DefaultTaskPlanExecutor.java:98)
	at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:63)
	at org.gradle.internal.concurrent.ManagedExecutorImpl$1.run(ManagedExecutorImpl.java:46)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at org.gradle.internal.concurrent.ThreadFactoryImpl$ManagedThreadRunnable.run(ThreadFactoryImpl.java:55)
	at java.lang.Thread.run(Thread.java:745)
Caused by: org.gradle.tooling.BuildException: Failed to process resources, see aapt output above for details.
	at com.android.build.gradle.internal.res.LinkApplicationAndroidResourcesTask.invokeAaptForSplit(LinkApplicationAndroidResourcesTask.java:512)
	at com.android.build.gradle.internal.res.LinkApplicationAndroidResourcesTask.doFullTaskAction(LinkApplicationAndroidResourcesTask.java:249)
	at com.android.build.gradle.internal.tasks.IncrementalTask.taskAction(IncrementalTask.java:106)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.gradle.internal.reflect.JavaMethod.invoke(JavaMethod.java:73)
	at org.gradle.api.internal.project.taskfactory.IncrementalTaskAction.doExecute(IncrementalTaskAction.java:46)
	at org.gradle.api.internal.project.taskfactory.StandardTaskAction.execute(StandardTaskAction.java:39)
	at org.gradle.api.internal.project.taskfactory.StandardTaskAction.execute(StandardTaskAction.java:26)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter$1.run(ExecuteActionsTaskExecuter.java:121)
	at org.gradle.internal.progress.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:336)
[... cut ...]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	... 1 more
Caused by: com.android.builder.internal.aapt.v2.Aapt2Exception: AAPT2 error: check logs for details
	at com.android.builder.png.AaptProcess$NotifierProcessOutput.handleOutput(AaptProcess.java:443)
	at com.android.builder.png.AaptProcess$NotifierProcessOutput.err(AaptProcess.java:395)
	at com.android.builder.png.AaptProcess$ProcessOutputFacade.err(AaptProcess.java:312)
	at com.android.utils.GrabProcessOutput$1.run(GrabProcessOutput.java:104)
```

[image]

sibelius, 2018-05-02:
disable appt2 for now, it is not supported by react-native yet

rexlow, 2018-05-02:
@sibelius I have disabled it, but another error pops up. Perhaps it has to do with the sdk version I'm on?

```
Process 'command '/Users/rexlow/Library/Android/sdk/build-tools/27.0.3/aapt'' finished with non-zero exit value 1
```

I am not sure which sdk version should I use (coming from iOS), here's the configuration.

```
    compileSdkVersion 23
    buildToolsVersion "23.0.1"

    ...

    dependencies {
        ...
        implementation "com.android.support:exifinterface:+" . <-- mixing versions can lead to runtime crashes
        implementation "com.android.support:appcompat-v7:23.0.1"
        ...
    }
```

sibelius, 2018-05-02:
follow this example https://github.com/react-native-community/rncamera-example

code-by, 2018-05-02:
Same error, my app/build.gradle

```
apply plugin: "com.android.application"

import com.android.build.OutputFile

project.ext.react = [
    entryFile: "index.js"
]

apply from: "../../node_modules/react-native/react.gradle"
def enableSeparateBuildPerCPUArchitecture = false
def enableProguardInReleaseBuilds = false

android {
    compileSdkVersion 23
    buildToolsVersion "23.0.1"

    defaultConfig {
        applicationId "com.rn055"
        minSdkVersion 16
        targetSdkVersion 22
        versionCode 1
        versionName "1.0"
        ndk {
            abiFilters "armeabi-v7a", "x86"
        }
    }
    splits {
        abi {
            reset()
            enable enableSeparateBuildPerCPUArchitecture
            universalApk false  // If true, also generate a universal APK
            include "armeabi-v7a", "x86"
        }
    }
    buildTypes {
        release {
            minifyEnabled enableProguardInReleaseBuilds
            proguardFiles getDefaultProguardFile("proguard-android.txt"), "proguard-rules.pro"
        }
    }
    // applicationVariants are e.g. debug, release
    applicationVariants.all { variant ->
        variant.outputs.each { output ->
            def versionCodes = ["armeabi-v7a":1, "x86":2]
            def abi = output.getFilter(OutputFile.ABI)
            if (abi != null) {  // null for the universal-debug, universal-release variants
                output.versionCodeOverride =
                        versionCodes.get(abi) * 1048576 + defaultConfig.versionCode
            }
        }
    }
}

dependencies {
    // compile project(':react-native-camera')
implementation (project(':react-native-camera')) {
        exclude group: "com.android.support"
    }
implementation "com.android.support:exifinterface:+"
  implementation "com.android.support:support-annotations:+"
  implementation "com.android.support:support-v4:+"

    compile fileTree(dir: "libs", include: ["*.jar"])
    compile "com.android.support:appcompat-v7:23.0.1"
    compile "com.facebook.react:react-native:+"
}

// Run this once to be able to run the application with BUCK
// puts all compile dependencies into folder libs for BUCK to use
task copyDownloadableDepsToLibs(type: Copy) {
    from configurations.compile
    into 'libs'
}
```

got this
```
Loading dependency graph, done.
warning: the transform cache was reset.
bundle: Writing bundle output to: /~/rn055/RN055/android/app/build/intermediates/assets/release/index.android.bundle
bundle: Done writing bundle output

/~/.gradle/caches/transforms-1/files-1.1/appcompat-v7-23.0.1.aar/63be421add3cb80ad7738b9c93904cf8/res/values/values.xml:113:5-69: AAPT: error: resource android:attr/fontVariationSettings not found.
    
/~/.gradle/caches/transforms-1/files-1.1/appcompat-v7-23.0.1.aar/63be421add3cb80ad7738b9c93904cf8/res/values/values.xml:113:5-69: AAPT: error: resource android:attr/ttcIndex not found.
    
/~/rn055/RN055/android/app/build/intermediates/incremental/mergeReleaseResources/merged.dir/values/values.xml:187: error: resource android:attr/fontVariationSettings not found.
/~/rn055/RN055/android/app/build/intermediates/incremental/mergeReleaseResources/merged.dir/values/values.xml:187: error: resource android:attr/ttcIndex not found.
error: failed linking references.

> Task :app:processReleaseResources
Failed to execute aapt
com.android.ide.common.process.ProcessException: Failed to execute aapt
```

rexlow, 2018-05-02:
@code-by Glad to know I am not alone. 

@sibelius The example you provided earlier was not helpful. I have tried linking manually before opening this issue. Please reopen this issue.

code-by, 2018-05-02:
@rexlow 
I have found solution
https://github.com/crosswalk-project/cordova-plugin-crosswalk-webview/issues/205#issuecomment-371669478
adding to end of build.gradle
```
configurations.all {
    resolutionStrategy {
        force 'com.android.support:support-v4:27.1.0'
    }
}
```

rexlow, 2018-05-02:
@code-by and @jgfidelis both your suggestions was very helpful! Am finally able to run after adding

`multiDexEnabled true`  and `implementation 'com.android.support:multidex:1.0.1'`

to the build.gradle file

closed 2018-05-02 by rexlow

ducpt2, 2018-05-03:
@sibelius i follow you suggest and can view camera, but scan barcode not work at all. I check many issue about barcode, but no luck? I think my issue is i update Android studio to new version.
Any help pls?

brickolicious, 2018-05-03:
build.gradle

multidex true
--------------------
  compile ("com.android.support:support-v4:26.0.1") {
        force = true //<-- force dependency resolution to 26.0.1 in my case 
    }

This resolved it for me

ducpt2, 2018-05-04:
@brickolicious do you test barcode scan? Does it work?

brickolicious, 2018-05-04:
I scan QR succesfully, haven't tried barcodes

Mian007, 2018-05-15:
i am facing the problem
**Execution failed for task ':react-native-camera:compileDebugJavaWithJavac'.**
 after add these two solutions to my app/build  dependecies section 
`    implementation 'com.android.support:multidex:1.0.1'; 
     compile ("com.android.support:support-v4:26.0.1") {
     force = true //<-- force dependency resolution to 26.0.1 in my case
     } `

kkusanagi, 2018-05-18:
Thanks to @jgfidelis . I just change as below and it work.

```
implementation (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
        exclude group: "com.android.support"
    }
```

AndreiCalazans, 2018-05-25:
A mix of things worked for me.
``` 
  react: 16.3.1 => 16.3.1
  react-native: 0.55.1 => 0.55.1
```

Add the following to your dependecies inside `android/app/build.gradle`
``` 
compile ("com.android.support:support-v4:26.0.1") {
        force = true //<-- force dependency resolution to 26.0.1 in my case
    }
```

Also make sure to update `android/build.gradle` to use gradle 3 and add google 

```  
// Top-level build file where you can add configuration options common to all sub-projects/modules.
buildscript {
    repositories {
        jcenter()
        google()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:3.1.0'
        
        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        mavenLocal()
        jcenter()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        google()
    }
}
```

Don't forget to also to update the distribuitionUrl in `android/gradle/wrapper/gradle-wrapper.properties` 

```
distributionUrl=https\://services.gradle.org/distributions/gradle-4.4-all.zip
```

saivamsid, 2018-06-19:
@AndreiCalazans Thanks! it worked.

pmiri, 2018-06-19:
I added these changes and can get the build to compile. But when launching the debugger, the app crashes on right as it starts up. Is there possibly an SDK mismatch or something?

youngjuning, 2018-11-02:
this can solve my trouble

```
implementation (project(':react-native-camera')) {
    // http://t.cn/EwxC1tr
    exclude group: "com.android.support"
}
```