react-native-camera/react-native-camera #1533: RCTCamera resets zoom when switching type

Tsiniloiv, 2018-05-02:
### Which implementation are you using

RCTCamera

### Steps to reproduce

On Android, have a RCTCamera view with the zoom prop set to read from this.state.zoom, then do

```
this.setState({
          zoom: [any level],
          type: [front type if current camera is back camera, back type if current camera is front camera],
        });
```

### Expected behaviour
The other camera should presumably open with the zoom set to whatever the zoom prop is.

### Actual behaviour
The other camera opens with no set zoom at all.

It works fine when the component is first mounted and such, but not afterwards. I've run into two thrown exceptions while experimenting with different approaches:

In some cases an exception is thrown because a setZoom() call tries to use a camera after it's been released.

When trying to call setZoom directly through React Native at the same time as setting the state, or as a callback parameter in the setState, it throws an exception because it failed to connect to the camera service. Directly calling setZoom through React Native when not switching camera types works fine.

The current and closest thing to working, at least insofar as it doesn't directly throw an exception, is the above example, which comes out like this with some logging:

```
05-02 15:33:30.482 1953-1953/com.appname D/zoom: CameraView setZoom called, setting to 30
05-02 15:33:30.482 1953-1953/com.appname D/zoom: RCTCamera setZoom called, setting cameraType 1 to 30
05-02 15:33:30.672 345-11616/? W/QCameraParameters: [PARM_DBG] zoom_level = 30
05-02 15:33:31.112 345-6416/? W/QCameraParameters: [PARM_DBG] zoom_level = 0
05-02 15:33:35.572 1953-1953/com.appname D/zoom: CameraView setZoom called, setting to 20
05-02 15:33:35.572 1953-1953/com.appname D/zoom: RCTCamera setZoom called, setting cameraType 2 to 20
05-02 15:33:35.912 345-12088/? W/QCameraParameters: [PARM_DBG] zoom_level = 20
05-02 15:33:36.312 345-31706/? W/QCameraParameters: [PARM_DBG] zoom_level = 0
```

### Environment
- **Node.js version**: 9.3
- **React Native version**: 0.55.2
- **React Native platform + platform version**: Android 6.0.1, API 23

### react-native-camera
**Version**: Master branch, but the RCTCamera parts are a modified version of 0.12, since RCTCamera saves pictures much faster than RNCamera, and has pinch zooming.

n1ru4l, 2018-09-24:
The initial zoom is currently not set, feel free to send a pull request that implements that feature!

SimonErm, 2018-11-21:
RCTCamera is deprecated

closed 2018-11-21 by n1ru4l