react-native-camera/react-native-camera #1534: How to use 'onGoogleVisionBarcodesDetected' on iOS

phuhk, 2018-05-02:
Hi there, I'm face on problems about scan QR code on iOS.
I'm using default props for RNCamera to scan QR code (not define specific barcodeTypes, just implement to scan all barcodeTypes).

In Android is work well, but on iOS I have crash on this function:
`- (void)captureOutput:(AVCaptureOutput *)captureOutput didOutputMetadataObjects:(NSArray *)metadataObjects
       fromConnection:(AVCaptureConnection *)connection`

metadataObjects stringValue is null.
I know Android is using Google Mobile Vision (GMV) to scan QR code, i see props 'onGoogleVisionBarcodesDetected', I want to ask how can I use GMV to scan QR code?

sibelius, 2018-05-02:
this was only implemented on Android side, we would accept a PR to add this feature on iOS

phuhk, 2018-12-03:
I must code native iOS using `ZXing` lib to resolve this case. Any one need help can pm me.

brayanL, 2019-03-21:
Really the BarCode functionality is not available for IOS..? you have any alternative..? :(

sibelius, 2019-03-21:
try using `onBarCodeRead` instead

brayanL, 2019-03-21:
Thank you very much @sibelius your last answer was a few confusion, but i understand that Google Vision for BarCode is only available for android, but IOS BarCode Scan works but without him.

sibelius, 2019-03-21:
can we close this?

brayanL, 2019-11-05:
onGoogleVisionBarcodesDetected is not available for IOS yet?

minhdang-idev, 2020-06-11:
This feature is really important. And why is it only implemented on android?

sibelius, 2020-06-11:
check latest version

closed 2020-06-18 by stale[bot]