react-native-camera/react-native-camera #1535: get supported preview sizes

code-by, 2018-05-02:
it is possible somehow get list of camera supported preview sizes?

sibelius, 2018-05-02:
yep

https://github.com/react-native-community/react-native-camera/blob/master/src/RNCamera.js#L246

code-by, 2018-05-02:
I got only aspect ratios, not dimensions, ex. 800x600

sibelius, 2018-05-02:
not sure this is possible

code-by, 2018-05-02:
I don't know is this still available for latest API, but found this
https://developer.android.com/reference/android/hardware/Camera.Parameters#getSupportedPreviewSizes()

sibelius, 2018-05-02:
what about in camera2?

we would accept a PR to expose this native method in js in RNCamera

code-by, 2018-05-02:
unfortunatly I'm not familiar with Android API (

code-by, 2018-05-02:
p.s. also need to detect current camera image size (or maximum supported size)

jonathangreco, 2018-11-15:
@sibelius Do I have to understand here, that preview camera is always in 4:3 ? Even if the capture is in 1080p ratio ?

IssuehuntBot, 2018-12-03:
@issuehuntfest has funded $20.00 to this issue. [See it on IssueHunt](https://issuehunt.io/repos/33218414/issues/1535)

sibelius, 2019-04-01:
related to this one https://github.com/react-native-community/react-native-camera/issues/940#issuecomment-478405048

sibelius, 2019-04-01:
you can get this using this method

https://github.com/react-native-community/react-native-camera/blob/ca35ac609d2e1d0b93f06056f3d1a78424e25dca/src/RNCamera.js#L400

`getAvailablePictureSizes`

check this example code

```jsx
onCameraReady = async () => {
    if (this.camera) {
      const ratios = await this.camera.getSupportedRatiosAsync();

      const sizes = await this.camera.getAvailablePictureSizes();

      console.log('onCameraReady ratios: ', ratios);
      console.log('onCameraReady sizes: ', sizes);
    }
  };
```

result:

```
'onCameraReady ratios: ', [ '3:2', '4:3', '5:3', '11:9', '16:9' ]
'onCameraReady sizes: ', [ '1280x720', '1920x1080' ]
```

it will only return sizes based on the current ratio

closed 2019-04-01 by sibelius