react-native-camera/react-native-camera #1536: Is it possible to take a picture before read the barcode?

andrebecklas, 2018-05-03:
I was minding if is it possible to automatically take a picture when the event onBarcodeRead is called, so it would return data, type and picture objects,

Thanks,

jgfidelis, 2018-05-03:
You could call `takePictureAsync` method on you onBarcodeRead event.

andrebecklas, 2018-05-07:
Is not working. I've tried to call both _this.takePicture.bind(this_) and _this.takePicture()_ in onBarCodeRead. When I take the photo it works. Nothing written in the log when a barcode is regonized. 

```
onBarCodeRead = e => {
    this.takePicture.bind(this);
    this.props.navigation.state.params.updateBarcode(e.data);
    this.props.navigation.goBack();
  };

  takePicture = async function() {
    if (this.camera) {
      const options = { width: 400, quality: 0.5, base64: true };
      const data = await this.camera.takePictureAsync(options);      
      console.log(data.base64);
    }
  };
```

andrebecklas, 2018-05-07:
I`ve tried to put the bind in properties also: _onBarCodeRead={this.onBarCodeRead.bind(this)}_

SimonErm, 2018-09-25:
Try this
```
onBarcodeRead = data => {
    if (this.state.barcode != data.data) {
      this.setState({ barcode: data.data }, this.takePicture);
    }
  };
  takePicture = async () => {
    if (!this.camera) return;
    const options = { width: 1080, base64: false, quality: 1 };
    const data = await this.camera.takePictureAsync(options);
    console.log(data);
  };
 render() {
    return (
      <View style={{ flex: 1 }}>
        <RNCamera
          ref={ref => (this.camera = ref)}
          style={styles.preview}
          onBarCodeRead={this.onBarcodeRead}
        />
      </View>
    );
  }
```

closed 2019-06-08 by stale[bot]