react-native-camera/react-native-camera #1537: Camera crashes after taking picture in Android

mykelaballe, 2018-05-03:
Implementation: RCTCamera

### Does it work with Expo Camera? I haven;t tested

### Expected behaviour
it shouldn't crash

### Actual behaviour
crashes after taking picture (i.e after the camera sound)

### Environment
Node: 9.10.1
React Native: 0.34
Platform: Android 4.2.2
Gradle: 1.3.1
React Native Camera: 0.4.1

Do I need to upgrade my environment versions? Coz right now I'm having trouble upgrading, like bumping up the gradle version causes the app to not compile at all

jgfidelis, 2018-05-03:
If you want to use RCTCamera, you could at least versions 0.13, or 1.0.0... Closing this issue because 0.4.1 is way too old

closed 2018-05-03 by jgfidelis

mykelaballe, 2018-05-03:
Thanks for the response. I'll try your suggestion