react-native-camera/react-native-camera #1540: The captured image is wrong orientation and take a long time for capturing

ducgao, 2018-05-04:
I'm using react-native-camera to implement snap feature. The screen I developed is fixed orientation so when I capture a image, the output is not rotate follow the right direction and the capturing process is take a long time.

I think react-native-camera need to use sensor manager to get the right orientation of device when capture instead of listen to orientation changed event.

The skipProcessing option didn't help a lot. The capture process still take a time to be done.
:(

jgfidelis, 2018-05-04:
Android still needs a lot of work. We need people to send PRs with these improvements.

Asalem1, 2018-09-18:
Have there been any developments on this @ducgao ?

closed 2019-06-08 by stale[bot]

ducgao, 2019-06-08:
I folked the repo and did it myself. if you feel interest, please visit my repo