react-native-camera/react-native-camera #1545: undefinde is not an object error after it's been linked

Voyager-One, 2018-05-07:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Gradle 3
This package required Gradle 3, follow this migration guide https://developer.android.com/studio/build/gradle-plugin-3-0-0-migration

### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. `yarn add react-native-camera`, `react-native link react-native-camera`
[image]

2. just import the RNCamera 
`import { RNCamera } from 'react-native-camera'`
[image]
So I think it is not actually linked
3. in xcode
[image]

[image]

### Does it work with Expo Camera?

expo camera works fine

### Expected behavior

successfully imported

### Actual behaviour

it looks like that it is not actually linked with the project. but the `react-native link` did give success message.

### Environment
- **Node.js version**: 10.0.0
- **React Native version**: 0.55.2
- **React Native platform + platform version**: iOS 11.3

### react-native-camera
**Version**: 1.1.2

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgfidelis, 2018-05-08:
I don't know what went wrong but that is a classical wrong link error...

jgfidelis, 2018-05-08:
Try doing the manual link

alopes, 2018-05-08:
I'm getting the same exact issue on a new project.

Voyager-One, 2018-05-18:
hey guys, I think I finally figure it out. if you are using expo (`create-react-native-app`) and want to use rn libs that needs to run `react-native link`, you must eject the expo project........

Mickamwamba, 2018-05-25:
Same issue here.. I am using 
react-native-cli: 2.0.1
react-native: 0.55.3
macOS Sierra v 10.12.6

JindiHu, 2018-06-19:
@Mickamwamba i thought this error happens in IOS. I dont think it related to android gradle. 
by the way, i also have the same problem with ios linking.

sibelius, 2019-04-18:
the link is wrong

closed 2019-04-18 by sibelius