react-native-camera/react-native-camera #1546: Quality prop for photo not working

VrajSolanki, 2018-05-07:
So I am trying to migrate to RNCamera from RCTCamera and there to control the quality we had CaptureQuality prop, but I guess I am doing something wrong because the quality prop in the options is not working for me. It is still taking the images in the highest resolution for both android and ios.

Please suggest what I might be doing wrong.

```
takePicture = async () => {
        if (this.cameraComponent) {
         const options = {  fixOrientation:true, forceUpOrientation:true, quality: 0.2};
          const data = await this.cameraComponent.takePictureAsync(options)
          console.log(data)
        }
      };
```

jgfidelis, 2018-05-07:
Can you try to compare the pictures sizes generated with quality 1.0 and quality 0.2?

This options specify the amount of compression done at the photo. 1.0 is basically not compression.

You are not doing anything wrong.

VrajSolanki, 2018-05-08:
Yes I tried everything 0.0, 0.1, 0.2 but it does not work. I am currently on version 1.1.0

n1ru4l, 2018-05-08:
@VrajSolanki Android or iOS?

VrajSolanki, 2018-05-08:
Faced in both of them.

n1ru4l, 2018-05-08:
@VrajSolanki It works for me on iOS, value `1` results in a file which is about `4mb` large and the value `0.1` results in a around `1.5mb` large picture. However I expect the file to be much smaller when using `0.1`..

jgfidelis, 2018-05-08:
You can look at the native code and see that it is used by native methods to compress the resulting image. Like @n1ru4l says, it clearly works as expected, making the resulting image smaller with smaller values for `quality`. I don't know what you expected or mean by "highest resolution".

closed 2018-05-08 by jgfidelis

n1ru4l, 2018-05-08:
Furthermore you can use additional packages for advanced compression. E.g. I am using my own written service for compressing the images to about 300kb which I was not able to do with the `quality` prop.

sibelius, 2018-05-08:
@n1ru4l can you share some code to apply some compression?

it would be good on recipes as well

n1ru4l, 2018-05-08:
@sibelius Sure here are my objective-c files:

```objc
//
//  MediaCompressionService.h
//
//  Created by Laurin Quast on 26.02.18.
//  Copyright © 2018 Laurin Quast. All rights reserved.
//

#import <React/RCTBridgeModule.h>
#import <AVFoundation/AVAsset.h>
#import <AVFoundation/AVAssetExportSession.h>
#import <UIKit/UIKit.h>

#ifndef MediaCompressionService_h
#define MediaCompressionService_h

@interface MediaCompressionService : NSObject <RCTBridgeModule>
@end

#endif /* MediaCompressionServicee_h */

```

```objc
//
//  MediaCompressionService.m
//
//  Created by Laurin Quast on 26.02.18.
//  Copyright © 2018 Laurin Quast. All rights reserved.
//

#import "MediaCompressionService.h"

@implementation MediaCompressionService

RCT_EXPORT_MODULE();

RCT_REMAP_METHOD(compressVideo,
                 inputFileUrl:(NSString *)inputFilePath
                 compressVideoResolver:(RCTPromiseResolveBlock)resolve
                 compressVideoRejecter:(RCTPromiseRejectBlock)reject)
{
  NSString *sanitizedInputFilePath = [inputFilePath stringByReplacingOccurrencesOfString:@"file:/" withString:@""];
  NSURL *inputFileUrl = [NSURL fileURLWithPath:sanitizedInputFilePath];
  NSURL *outputFileUrl = [self createTemporaryFileUrlWithExtension:@"mov"];
  AVURLAsset *urlAsset = [AVURLAsset URLAssetWithURL:inputFileUrl options:nil];
  AVAssetExportSession *exportSession = [[AVAssetExportSession alloc] initWithAsset:urlAsset presetName:AVAssetExportPresetMediumQuality];
  exportSession.outputURL = outputFileUrl;
  exportSession.outputFileType = AVFileTypeQuickTimeMovie;
  exportSession.shouldOptimizeForNetworkUse = YES;
  [exportSession exportAsynchronouslyWithCompletionHandler:^{
    if (exportSession.error) {
      reject(@"error", @"compress video", exportSession.error);
      return;
    }
    resolve([outputFileUrl absoluteString]);
  }];
}

RCT_REMAP_METHOD(compressImage,
                 inputFileUrl:(NSString *)inputFilePath
                 compressImageResolver:(RCTPromiseResolveBlock)resolve
                 compressImageRejecter:(RCTPromiseRejectBlock)reject)
{
  NSString *sanitizedInputFilePath = [inputFilePath stringByReplacingOccurrencesOfString:@"file:/" withString:@""];
  NSURL *outputFileUrl = [self createTemporaryFileUrlWithExtension:@"jpeg"];

  NSData *imageData = [[NSFileManager defaultManager] contentsAtPath:sanitizedInputFilePath];
  
  CGFloat scale= (100*1024)/(CGFloat)[imageData length];
  UIImage *image = [UIImage imageWithData:imageData];
  UIImage *small_image=[UIImage imageWithCGImage:image.CGImage scale:scale orientation:image.imageOrientation];
  imageData = UIImageJPEGRepresentation(small_image, scale*1.00);

  [imageData writeToURL:outputFileUrl atomically:NO];
  resolve([outputFileUrl absoluteString]);
}

- (NSURL*)createTemporaryFileUrlWithExtension:(NSString *) extension {
  NSString *tempFileName = [NSString stringWithFormat:@"%@.%@", [[NSUUID new] UUIDString], extension];
  NSString *tempDirectory = NSTemporaryDirectory();
  NSString *tempPath = [tempDirectory stringByAppendingPathComponent:tempFileName];
  return [NSURL fileURLWithPath:tempPath];
}

@end

```

**Usage Example**

```js
//@flow

interface MediaCompressionService = {
  compressVideo: (path: string) => Promise<string>,
  compressImage: (path: string) => Promise<string>,
}

const {
  MediaCompressionService,
}: { MediaCompressionService: MediaCompressionService } = NativeModules

MediaCompressionService
  .compressImage(imagePath)
  .then(compressedFilePath => console.log(compressedFilePath))
MediaCompressionService
  .compressImage(videoPath)
  .then(compressedFilePath => console.log(compressedFilePath))
```

No android implementation yet 👽

jgfidelis, 2018-05-08:
that looks like a nice bridge to release if you want to

VrajSolanki, 2018-05-09:
@n1ru4l thanks alot.
@jgfidelis Yes that is what is happening for me a 5mb image is getting to 2mb but what happens is we expect the item to compress greatly from 1.0 to 0.1 so I understood that 5mb image get compressed to 500kb.
Anyways my bad in understanding, I guess will have to use custom implementation to further reduce the size.
Seems like an interesting discussion for the folks.

n1ru4l, 2018-05-09:
@jgfidelis  @VrajSolanki Maybe we should implement an option to further compress the video?

I did no see any more improvements by using values lower than `0.1` e.g. (`0.0005`). Firthermore the value `0` which is also valid (according https://developer.apple.com/documentation/uikit/1624115-uiimagejpegrepresentation?language=objc) results in no compression, seems like it is interpreted as `undefined` or `null`.

VrajSolanki, 2018-05-09:
I feel that would be helpful and the user would not have to compress it using another library, since we are able to compress it to say 1.5mb i believe doing it further would not be difficult. 
Having said that this is the pitfall I faced for my use-case. 
Pondering, is it a good thing to have? @n1ru4l @sibelius