react-native-camera/react-native-camera #1549: onTextRecognized never callback.

ribuluo000, 2018-05-08:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Gradle 3
This package required Gradle 3, follow this migration guide https://developer.android.com/studio/build/gradle-plugin-3-0-0-migration

gradle v4.4
### Which implementation are you using

RNCamera

### Steps to reproduce
1.Start a new project,and add react-native-camera to your project. The camera can take photo correctly.
2.
App.js
```

import React, { Component } from 'react';
import {
    StyleSheet,
    Text,
    TouchableOpacity,
    View
} from 'react-native';
import { RNCamera } from 'react-native-camera';

type Props = {};
export default class App extends Component<Props> {
  render() {
      return (
          <View style={styles.container}>
            <RNCamera
                ref={ref => {
                    this.camera = ref;
                }}
                style = {styles.preview}
                type={RNCamera.Constants.Type.back}
                flashMode={RNCamera.Constants.FlashMode.on}
                permissionDialogTitle={'Permission to use camera'}
                permissionDialogMessage={'We need your permission to use your camera phone'}
                // onFacesDetected={(d)=>{console.log('onFacesDetected',d);}}
                // onFaceDetectionError={(d)=>{console.log('onFaceDetectionError',d);}}
                onTextRecognized={(d)=>{
                  console.log('onTextRecognized',d);
                  alert(d);

                }}
            />
            <View style={{flex: 0, flexDirection: 'row', justifyContent: 'center',}}>
              <TouchableOpacity
                  onPress={this.takePicture.bind(this)}
                  style = {styles.capture}
              >
                <Text style={{fontSize: 14}}> SNAP </Text>
              </TouchableOpacity>
            </View>
          </View>
      );
  }

    takePicture = async function() {
        if (this.camera) {
            const options = { quality: 0.5, base64: true };
            const data = await this.camera.takePictureAsync(options)
            console.log(data.uri);
        }
    };

}

const styles = StyleSheet.create({
    container: {
        flex: 1,
        flexDirection: 'column',
        backgroundColor: 'black'
    },
    preview: {
        flex: 1,
        justifyContent: 'flex-end',
        alignItems: 'center'
    },
    capture: {
        flex: 0,
        backgroundColor: '#fff',
        borderRadius: 5,
        padding: 15,
        paddingHorizontal: 20,
        alignSelf: 'center',
        margin: 20
    }
});

```

3.
onTextRecognized never callback.

### Does it work with Expo Camera?
No.

RNCamera is based on Expo Camera, if this does not work with Expo Camera it won't work with RNCamera as well.
You should open an issue there as well, so we can cooperate in a solution.

### Expected behaviour
onTextRecognized should callback.

### Actual behaviour
onTextRecognized never callback.

### Environment
- **Node.js version**:v8.1.3
- **React Native version**:v0.55.3
- **React Native platform + platform version**: Android 5.0

### react-native-camera
**Version**: npm version v1.1.2

jgfidelis, 2018-05-08:
Is the Native code detecting the Text and not sending the event to the JS component?

sibelius, 2019-04-18:
try on latest version 

it uses MLKit now

closed 2019-04-18 by sibelius

ShahryarAhmed97, 2020-05-18:
You have made a spelling mistake in function name ->onTextRecognized
it should be like this ->onTextRecognize